soar-telescope/goodman_pipeline #1: Improve function to find targets

simontorres, 2016-07-21:
The module **process.py** contains a class Process.

The method _identify_spectra_ is not written in an elegant way thus is a bit hard to read.

simontorres, 2016-11-29:
started a set of simulation tools in order to identify the most conflictive steps/conditions and hopefully improve the extraction

simontorres, 2016-11-29:
Detection works nicely, extraction needs focus.

simontorres, 2016-11-29:
I might use what @b1quint described in #4 
```python
import astropy.io.fits as pyfits
import numpy as np
from scipy import signal
spec = pyfits.getdata(filename)
foo = np.where(np.abs(spec > spec.min() + 0.75 * spec.max()), spec, 0)
maxima = signal.argrelmax(foo, axis=0, order=10)[0]
```
If I can make it work for a single peak.

b1quint, 2016-12-07:
The 'signal.argrelmax' does not work well on 2D data. It is easier (?) to split each column in a 1D array and work one by one. If needed, you can use 'map' or 'multiprocessing.Pool.map_async'. Here is an example:
'
x = (some 2D array)
order = int(0.25 * x.shape[0])
find_peak = lambda i: signal.argrelmax(x[i], axis=0, order=order)[0]
peaks = map(find_peak, numpy.arange(x.shape[1]))
'

simontorres, 2018-02-20:
It has been done.

closed 2018-02-20 by simontorres