soar-telescope/goodman_pipeline #4: Improve line center finding for the spectrum being calibrated (lamp)

simontorres, 2016-10-05:
Right now I'm using the barycentre of a region of a line to select the best approximate centre of a line. The region is selected by mouse input but this is not always precise.
My idea is to do the following:

1. Given a mouse input (click) find the nearest peak
2. From the peak to the left and to the right, find where it stops decreasing (we are looking at emission lines). This will be a cleaner region to find the new barycentre. 
3. Alternatives are to fit a profile (from experience: not good) or just use the peak.

b1quint, 2016-10-07:
Did you try `astropy.modeling.models` and `astropy.modeling.fitting`? They seem to me to be pretty robust and easy to use. Even when your line has more than one component.

simontorres, 2016-10-07:
I've tried it, but I wasn't satisfied with the result. The centroids gave better results.

simontorres, 2016-11-11:
I'm trying a different method, maybe less elegant but way more effective. The method goes through all data points in a 1-D spectrum and keep track of the pixel value and whether there is a rising trend or falling trend. If the data is in a rising trend and a fall is detected it will check that the length of the trend is at least three pixels and if the data value is above the median. If those criterium are met the point will be considered a line candidate. Further improvement can be done but I say we leave it as it is since a more complicated algorithm might consume more resources.

simontorres, 2016-11-17:
Moffat does not improve result. Will continue using Voigt

simontorres, 2016-11-22:
Added a centering method. is not perfect but the cost of improving it versus the gain makes it not worth doing.

b1quint, 2016-11-24:
In order to find peaks you may also have a look on `scipy.signal.argrelmax`. This method allows you to do more or less the same thing that you described above. It also let you choose what axis you are looking at and what is the size of the window for comparison.

Here is a good example:
http://stackoverflow.com/a/13491866/2333908

b1quint, 2016-11-24:
For data with SAMFP I had a similar case and, in order to select only the brightest lines I used the following piece of code:

> import astropy.io.fits as pyfits
> import numpy as np
> spec = pyfits.getdata(filename)
> foo = np.where(np.abs(spec > spec.min() + 0.75 * spec.max()), spec, 0)
> maxima = signal.argrelmax(foo, axis=0, order=10)[0]

The `[0]` is needed because `argrelmax` returns a tuple.

b1quint, 2016-11-24:
Once you have the peak, gather the slit size on the header and transform it from arcseconds to pixels and calculate the barycenter within (a fraction of) this window.

simontorres, 2016-11-29:
I have made some tests using similar parameters but it doesn't improve greatly what I have running at the moment, I would definitely include this after the first release in order to improve efficiency and to have cleaner code. I should study more the parameters and test them.

simontorres, 2017-03-07:
Finally I implemented this, but now it can't detect the (very broad) lines that creates a 5" slit. Could use some help from @b1quint

b1quint, 2017-03-08:
I believe that the algorithm should be different for different sizes of slits. For example `argrelmax` works very well if we have a well defined local maximum and it works with several maxima.

For broad lines, I suggest having two interactions:
1) Try to find local maxima using `argrelmax` with order calculated using the slit size as input (in case we have this information in the header). Do we actually have the slit size in the header? 

2) Estimate the center of the line:
a) By fitting a gaussian multiplied/convolved to a square function. Again the size of the slit is need.
b) Estimate the barycenter of the line using a window based on some criterium like the slit size or the data that is descending from the center found using `argrelmax`.

simontorres, 2017-03-08:
I agree, currently the order is calculated using the slit size, which can be extracted from the header. But `agrelmax` doesn`t detect any line.

I will try using the old method for slit sizes larger or equal than 5. Then we could ajust to smaller sizes if necessary.

simontorres, 2017-03-28:
I think line detection works quite well, if there problems in the future a new issue must be opened.

b1quint, 2017-03-29:
What was actually the solution used? And what metrics did you use to determine that it is running well?

simontorres, 2017-03-29:
I implemented the code you suggested plus using slit and binning value to estimate the size of the `order` parameter which has to do with the width of lines to be detected. One thing I didn't cover was how to automatically set the percentage of the maximum value to be used as limit, so far it doesn't seems to be a problem but I've done external (to the pipeline) tests in which the line detection depends on that value, for instance:
```python
foo = np.where(np.abs(spec > spec.min() + 0.75 * spec.max()), spec, 0)
```
In the previous _code sentence_ `0.75` would be the value to automatize.

I have no formal metrics. How would you do it? because for me is just a visual inspection on the amount of potential useful lines detected. At the moment I don't know a method to measure this automatically. But I'll keep this in mind.

simontorres, 2018-02-20:
Long story short: I've tried several method and, at least for narrow slits, the _peak finding routines_ give the best results already (first approach)

closed 2018-02-20 by simontorres