soar-telescope/goodman_pipeline #6: Do automatic wavelength solution

simontorres, 2016-10-06:
Interactive wavelength solution is working very well, but we need the software to work mostly automatic.

b1quint, 2016-10-07:
What are the goal errors at the moment? From the e-mails, I understand that you can get RMS < 0.1 A with Interactive Mode (how many points? ) and RMS <  1.0 A with Automatic Mode.

simontorres, 2016-10-07:
The best I've reached so far is 0.06A in interactive mode, combining manual selection of points then automaticly adding more points and at the end visual examination to remove worst points. I should add a points counting for the solution but I would say with 6 you already have a solution with RMSE of about 1-2A (if the points are widely spread across the full spectrum) and the function that adds more points add 20 or so.

Cesar said: if interactively I can reach 0.06A in automatic mode I should expect at least 0.6A.

b1quint, 2016-10-07:
Perfect. Thanks for clarifying. Anyways, what if 0.1 A and 1 A RMS for Interactive Mode and Automatic Mode should be set as goal? If so, you would be already doing good and that is a reliable number that we can give to the user. If you promise 0.06A for Interactive Mode, for example, but that happened just once, the user can be frustrated to get any little worse than that (e.g., 0.07 A). Maybe this "constraint relaxation" should not be set right now, but in the future when you have more clear what are the average RMS for each mode after several trials.

simontorres, 2016-11-09:
Did a first approach of obtaining a sintetic template for a given lamp using NIST linelist with relative intensity but the result are too far away from reality, my guess is that it has to do with the conditions in which NIST linelist are produced and the conditions in which comparison lamps operate. The most likely path will be to build templates using the same lamps.

closed 2017-04-05 by simontorres