soar-telescope/goodman_pipeline #13: Handle multiobjects

simontorres, 2016-11-15:
Multi objects can be identified but there is no handling for them

simontorres, 2016-11-22:
There a preliminary implementation of this. Need to run a few tests and probably improve the extraction method in order to avoid contamination or cross-contamination of sources.
The name of the output files for multiple targets will be: __original_filename_1.fits__, __original_filename_2.fits__ ... __original_filename_N.fits__ and for single ones will be simply __original_filename.fits__ also consider the __g__  as prefix.

b1quint, 2016-11-24:
It would be good to have a performance parameter for object separation. Remember that you will not be able to detect different sources if they are too close. People say that they can not be resolved. That depends on their separation and on the observed seeing. Considering that the spacial fwhm is the seeing, they normally can only be resolved if they are separated by more than a fwhm / 2.

simontorres, 2018-05-17:
We will simulate data with more than one target and see how the identification works and then do the same for extraction.

simontorres, 2018-11-06:
This issue is not specific enough and is very old.
See #258

closed 2018-11-06 by simontorres