soar-telescope/goodman_pipeline #14: Automatic Wavelength Calibration using templates.

simontorres, 2016-11-16:
1. Build lamp templates for every "spectroscopic mode" supported by our pipeline.
2. put them in one folder (all fits files, maybe not?)
3. Use ImageFileCollection + Pandas to do the selection according to certain parameters.  
  1. OBJECT (which lamps are in use)
  2. GRATING 
  3. MODE (if available)
  4. FILTER(s)
  5. etc

simontorres, 2017-04-04:
# TODO list to work on the automatic wavelength calibration (experimental) module

1.- Review @b1quint calculations for the grating equation also check UNC's work.
2.- Apply a global cross correlation for a first approach and then apply _local_ cross correlations to finetune the solution
3.- Adjust _fwhm_ by the size of the slit for the Gaussian convolution before the cross correlation.

simontorres, 2017-04-05:
This issue replaces #6

simontorres, 2017-04-12:
# Note
Doing a global cross correlation doesn't help if the reference lamps are in the same configuration.

simontorres, 2017-04-17:
I completed a preliminary automatic wavelength solution, as a general description of what it does see the list.

- Loads a suitable template if none returns None (will skip that target)
- Find lines in new comparison lamp
- For each line find shift by doing _local_ cross correlation and stores it. Also stores pixel value and angstrom value
- Does a sigma clip of cross correlation values (shifts)
- Creates a wavelength solution by fitting the angstrom and pixel values.
- Using the new wavelength solution calculates the differences in angstrom to the reference values.
- Does sigma clip to the difrerences and removes the corresponding pixel and angstrom values.
- Creates a new and final wavelength solution. If there are no values clipped-out the first solution will be the final.

# Results:
It works well for similar configurations with RMS of the order of 0.4 Angstrom for the 400M2 configuration using the slit of 1.03 arcseconds.
It does not work for the slit of 5 arcseconds using the template taken with 1.03 arcseconds. It will produce a wavelength solution though but the RMS is of the order of 9.

One idea to make it work for the slit of 5 arcseconds is to convolve the template with and appropriate model and also scale it in the y-axis (intensity) given that there is no adding of shifts in the x-axis.

simontorres, 2017-04-18:
I implemented a Box1DKernel convolution and y-axis scaling to imitate the shape of the wider slit lines from a template taken with a much narrower slit (1.03) The RMS Error dropped from ~9 to 0.2. There will be little improvements to do but this is getting closer. See the following picture.
[image]

cbaorion, 2017-04-18:
> auto-wavelength2

Buenisimo!

Cesar

-- 
***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************