soar-telescope/goodman_pipeline #24: Document Error that causes inconsistencies in ImageFileColection tables.

simontorres, 2016-12-19:
```shell
Traceback (most recent call last):
  File "/user/simon/development/soar/goodman/redspec.py", line 807, in <module>
    MAIN_APP = MainApp()
  File "/user/simon/development/soar/goodman/redspec.py", line 54, in __init__
    self.night = self.set_night()
  File "/user/simon/development/soar/goodman/redspec.py", line 330, in set_night
    image_collection = ccd.ImageFileCollection(self.args.source, keys)
  File "/usr/lib/python2.7/site-packages/ccdproc-1.1.dev899-py2.7.egg/ccdproc/image_collection.py", line 103, in __init__
    self.keywords = keywords
  File "/usr/lib/python2.7/site-packages/ccdproc-1.1.dev899-py2.7.egg/ccdproc/image_collection.py", line 200, in keywords
    self._summary_info = self._fits_summary(header_keywords=new_keys)
  File "/usr/lib/python2.7/site-packages/ccdproc-1.1.dev899-py2.7.egg/ccdproc/image_collection.py", line 431, in _fits_summary
    summary_table = Table(summary_dict, masked=True)
  File "/usr/lib64/python2.7/site-packages/astropy/table/table.py", line 371, in __init__
    init_func(data, names, dtype, n_cols, copy)
  File "/usr/lib64/python2.7/site-packages/astropy/table/table.py", line 670, in _init_from_dict
    self._init_from_list(data_list, names, dtype, n_cols, copy)
  File "/usr/lib64/python2.7/site-packages/astropy/table/table.py", line 635, in _init_from_list
    self._init_from_cols(cols)
  File "/usr/lib64/python2.7/site-packages/astropy/table/table.py", line 700, in _init_from_cols
    .format(lengths))
ValueError: Inconsistent data column lengths: set([82, 164])

Process finished with exit code 1
```

simontorres, 2016-12-19:
I has to do with duplicated keywords. In particular
ENVTEM
ENVDIR

simontorres, 2017-03-07:
There is a method to solve this problem for the CCD and Spectroscopic pipeline

closed 2017-03-07 by simontorres

b1quint, 2017-03-29:
Just for information, what would be the name of the method?

simontorres, 2017-03-29:
The method is `goodman.goodman_ccd.data_classifier.fix_duplicated_keywords(night_dir)`
It modifies the header of the file so this runs only once.