soar-telescope/goodman_pipeline #27: Imaging Mode

simontorres, 2017-01-20:
The ccd reduction pipeline does not discriminate imaging filters. Therefore it tries to combine all flats _without grating_ in a single image.

cbaorion, 2017-01-20:
Claramente hay que modificar el pipeline para que use una opcion imaging 
o una espectroscopica. Si es la segunda, ejecuta la reduccion como ya lo 
viene haciendo.

La otra opcion que puede ser mas sencilla, es simplemente tener DOS 
pipelines, uno para reducir imagenes espectroscopicas, y otro para 
imaging. La reduccion inicial entonces puede ser muy parecida, solo que 
el pipeline de imaging organizara la data por el keyword FILTER, y 
ademas chequeara que GRATING=="<NO GRATING>".  Si GRATING!="<NO 
GRATING>" entonces deberia para y dar un error message.

O algo asi.

Cesar

-- 
***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************

simontorres, 2017-01-20:
Mi idea es hacer que si encuentra imagenes no-grating las separe por filtros. Tal vez encontremos mas errores mas adelante, pero dos pipelines me parece mas complicado que adaptar la ya existente.

cbaorion, 2017-01-20:
Ok. Eso suena bien.

simontorres, 2017-01-20:
Consulta: En imaging mode se usan ambas ruedas de filtros? o solo la primera y la segunda para los filtros de espectroscopia? o no existe ninguna regla? Es importante saber para poder programar esta parte. Otra cosa que estoy pensando es tener solo una pipeline pero que tenga un imaging y un spectroscopic mode, podria hacerse transparente al usuario (un poco mas complejo) o que el usuario elija cada vez. De cualquier modo aun necesito familiarizarme un poco mas con el codigo de reduccion de CCD para proponer la mejor solucion (segun mi criterio).

simontorres, 2017-03-07:
Issue is no longer valid since the new version of the CCD Reduction Pipeline solves it

closed 2017-03-07 by simontorres

b1quint, 2017-03-29:
I believe that the criterium for Imaging Mode Vs Spectroscopic Mode should be the grating angle and the camera angle. When we have both of them near to 0 deg, then we are at the imaging mode. Else, we have Spectroscopic mode. That does not relies on filters.

simontorres, 2017-03-29:
For that we would need to provide an image representative of the  night to the pipeline, as an argument, right now it selects a random image from the whole night. Here is the method that finds out whether is spectroscopy or imaging. I call it _observing technique_ . The instrument is determined in a similar way but one step before.

```python

```    def get_obs_technique(self):
        """Identify if the data is Imaging or Spectroscopy

        Besides the fact there are two cameras there are two observational techniques. Imaging and Spectroscopy. The
        red camera has an specific keyword that contains that information but the blue does not.

        The result is stored as an attribute of the class.

        """

        if self.instrument == 'Red':
            wavmodes = self.objects_collection.wavmode.unique()
            if len(wavmodes) == 1 and wavmodes[0] == 'Imaging':
                self.technique = 'Imaging'
                log.info('Detected Imaging Data from RED Camera')
            elif 'Imaging' in wavmodes and len(wavmodes) > 1:
                log.error('There are mixed observation techniques this night. Please classify your data')
                self.technique = 'Unknown'
            else:
                self.technique = 'Spectroscopy'
                log.info('Detected Spectroscopy Data from RED Camera')
        elif self.instrument == 'Blue':
            self.remove_conflictive_keywords()
            gratings = self.objects_collection.grating.unique()
            if gratings != ['<NO GRATING>']:
                self.technique = 'Spectroscopy'
                log.info('Detected Spectroscopy Data from BLUE Camera')
            elif gratings == ['<NO GRATING>'] and int(np.mean(self.objects_collection.cam_targ.unique())) == 0:
                self.technique = 'Imaging'
                log.info('Detected Imaging Data from BLUE Camera')
            else:
                log.error('It was not possible to determine observing technique')
                self.technique = 'Unknown'

b1quint, 2017-03-29:
I see. So if in one night we have both Imaging Mode and Spectroscopic Mode, the user will have to have these data separated before, right?

simontorres, 2017-03-29:
In principle yes. I would add the ability to do both, but again, that might consume us a lot of time.