soar-telescope/goodman_pipeline #28: CCD reduction software does not discriminate different gratings

simontorres, 2017-01-23:
When processing multi-grating data the ccd reduction pipeline can not distinguish different grating/filters configuration for flat correction. Also flats are not normalized.

simontorres, 2017-01-30:
Grating and filter discrimination solved. Need to check the flat normalization issue @b1quint

b1quint, 2017-01-31:
What would be the Flat Normalization Issue? I tried to find it but I could not. Could you cite it, give me the link or explain a bit better?

simontorres, 2017-01-31:
This is the "intensity profile" of the flat image.
[image]
[image]

simontorres, 2017-01-31:
I meant "master flat".

simontorres, 2017-03-07:
This issue is no longer valid since the updated version of the CCD Reduction pipeline

closed 2017-03-07 by simontorres