soar-telescope/goodman_pipeline #34: `bin` scripts relies on directory tree acquitecture

b1quint, 2017-02-01:
I have now the Goodman pipeline in my Mac OSX and I have all the dependencies but when I try to run the `redccd` script, I receive the following error:

```
$ redccd
Traceback (most recent call last):
  File "../../../bin/redccd", line 5, in <module>
    import goodman_ccd
ImportError: No module named goodman_ccd
```

This is caused because the script cannot find the `goodman_ccd` file or package. To fix that, I have to run this inside the project's root folder. That should not be like that. It is desirable that the script can be run from any directory. 

The best option is simply creating a `setup.py` file and use `distutils` to create an installation file. Of course, after each update, the user will have to re-install the package.

simontorres, 2017-02-01:
I'm aware of that problem but decided to focus on the documentation problems and the integration of the ccd reduction tool as part of the same package. The only missing variables to make it work are PYTHONPATH and adding the root directory to the systems' PATH. I will work on the distutils installation soon.

simontorres, 2017-02-02:
I succesfully created a package distribution with distutils but there is a small problem regarding the location of reference data and I will require your opinion so I will create a new issue.

simontorres, 2017-03-07:
This has been solved since creation of the package to distribute the pipeline

closed 2017-03-07 by simontorres