soar-telescope/goodman_pipeline #35: Regarding location of reference data for wavelength calibration

simontorres, 2017-02-02:
The wavelength calibration pipeline makes use of a set of previously calibrated spectrum in order to use them as visual reference when finding the match of spectral lines. 
Using `distutils` does create a problem. There are two ways to distribute this data according to [this site](http://stackoverflow.com/questions/4519127/setuptools-package-data-folder-location) (see best answer) which are:

1. Install as package data
    Installing as package data would be simpler but a normal user might find troubles adding new reference files.
2. Install to fixed location
    Installing to fixed location might be a little bit trickier to implement but it would be simpler for a user to add new reference files.

I would like to know your opinion on this regard.

Also see Issue #34

simontorres, 2017-02-06:
I finally used the option 1.
Installed as packaged data. Basically because it would be a most robust solution and since I also can provide a different path for calibration data in which case it would work as in the case 2.

b1quint, 2017-03-05:
Option 1 is usually better. 

From what I see we could close this issue?

simontorres, 2017-03-07:
It was done that way.
Closing Issue now.

closed 2017-03-07 by simontorres