soar-telescope/goodman_pipeline #37: What about fringing?

b1quint, 2017-03-05:
Recently a couple of astronomers asked me about calibration data and how to remove fringing using Goodman. @tribeiro told me that he wrote a guide for removing fringing. I looked at SOAR's webpages and could not find anything. I did not find that in my e-mails either. Do we actually have a suggested step by step reduction to remove fringing? How about implementing it?

cbaorion, 2017-03-05:
That may be useful for later on, and no, we do not provide specifics on 
how to deal with that. Tiago did his own method, and had a script for 
it. But I want to see the software actually working through a full dataset.

Also, its a secondary thing now that we have the Red Camera, which has 
almost no fringing.

Cesar

-- 
***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************

tribeiro, 2017-03-05:
What I did was to implement the technique developed by Paul Goudfrooij and Stefi A. Baum [http://www.stsci.edu/~stefano/newcal97/pdf/goudfrooijp2.pdf](url). You can find a description of the method for GOODMAN data here [https://docs.google.com/document/d/10lFYixbOwGlzGqUZEEMyaCFnh5BgpzO1xJ9-83R5ifE/edit?usp=sharing](url)

b1quint, 2017-03-15:
Just for reference, let me upload the files here since the links above not always work:

[Goodman - Remove Fringing from data.pdf](https://github.com/simontorres/goodman/files/844493/Goodman.-.Remove.Fringing.from.data.pdf)

[Fringe Correction for STIS.pdf](https://github.com/simontorres/goodman/files/844494/Fringe.Correction.for.STIS.pdf)

simontorres, 2020-12-01:
I haven't received a complaint about fringing, I'm closing this issue because is too old.

closed 2020-12-01 by simontorres