soar-telescope/goodman_pipeline #38: Uniform data reading mode

simontorres, 2017-03-14:
Change astropy.io.fits to ccdproc.CCDData

simontorres, 2017-03-14:
There is a bug that causes all data to be NaN value. Some of the fits reading methods is not dealing correctly with NaN. First I will change the data reading library and if that doesn't work will have to created a method to deal with NaNs. As far as I know ccdproc.CCDData deals better with them. Also ccdproc.convolution.convolve can deal with them.

simontorres, 2017-03-24:
NaNs are located outside the useful region. see image. I will try changing the trim region.
[image]

simontorres, 2017-03-27:
Changing the trim section does solve the problem, in other words, and as I said before, the NaN and -INF values are outside the useful region (or outside the slit image) so trimming to the limit, even the last column solves the problem but I would like a more robust solution, one that can handle **the right way** the NaNs or -INFs that may appear in the data region.

simontorres, 2017-03-28:
# Note
_The method to read the fits files has zero influence on the problem_

b1quint, 2017-03-29:
A simple solution is simply to substitute NaNs and INFs for a given finite constant number. That would be a quick-and-dirty solution. For example: you can simple substitute them all for -1e6 but this could cause other bugs in the later parts of the data-reduction (like mean or mode estimation).

simontorres, 2017-03-29:
I agree, and given that the NaNs and INFs are always outside the data area there would not be an impact but I think is possible that they may appear on the data zone and we need a robust method to deal with them. I think an interpolation would be the best, more time consuming but the result will be better.

simontorres, 2017-04-05:
The existance of NaN is always outside the interesting area for science but are so annoying and usually the whole data ends up converted in a collection of NaNs.

My definite solution would be to apply an interpolation in order to remove them.
This process will be right after the cosmic ray rejection and will be the last before writing the file.

simontorres, 2017-04-07:
I have tested a method to do cubic/linear interpolation but it takes too much time to process. We should evaluate using `numpy.nan_to_num()` which converts NaN to Zero, -INF to a very small number and INF to a very large number.

b1quint, 2017-04-10:
It that does not interfere in future operations, it would be a good approach.

simontorres, 2017-04-10:
I created an argument `--interpolate-invalid` that when parsed will do the interpolation, most likely we will have to review the process since it very power consuming. By default it will do the `nan_to_num` operation.

simontorres, 2017-04-11:
The method `numpy.nan_to_num()` is not good for this case. Apparently, since -INF and INF are converted into very large numbers it creates other problems that I will not look much into, unless the third alternative does not work.

The _third_ alternative is to convolve the image with what and how still don't know. I will use as a guide [this link](http://docs.astropy.org/en/stable/convolution/).

simontorres, 2017-04-17:
The solution for this was to manually define the trim section. It will be necessary to calculate the new one for the other supported binning modes.

simontorres, 2017-04-27:
I removed the argument `--interpolate-invalid` because the methods I tried were highly inefficient. And the solution to this issue was to trim the image a bit further to avoid the conflictive regions. Also I trimmed to the slit location which helped to avoid even more problems. (see Issue #57 )

closed 2017-04-27 by simontorres