soar-telescope/goodman_pipeline #41: Add possibility to manually select Extraction zones

simontorres, 2017-03-14:
In some very special cases there might happen that the automatic target identification and extraction will not work properly, so there should exist the possibility to, by parsing an argument (most likely), manually select the regions for the target and for background subtraction.
This could be particularly helpful for targets that are too close or too close to the border (nobody should put their science target near the border though).

b1quint, 2017-03-15:
Isn't this related to #1?

simontorres, 2017-03-15:
Absolutely, but I think this can be treated independently because one thing is to improve the target detection (automatically) and another is to add a manual option (similar to the interactive mode for wavelength calibration). Despite this I would consider merging both issues. By the way, this came out from the discussion after the talk in the SOAR 2020 Workshop.

b1quint, 2017-03-15:
This is one of the reasons that it has to be considered for the object extraction: if an input from the user may be an option, we have to have the code written to make this available in the future. If not, it may be too entangled so adding this new capability will be much more difficult.

cbaorion, 2017-03-29:
Yes, we **definitely** want this feature!  Simon should code things so that this can be added later, but I would not include it for the Beta release.

simontorres, 2017-03-29:
It will require some time, since we need to program a graphic user interface for this but if we are planning to create a special library for graphic treatment (see #42 )we better leave the effort for then.

simontorres, 2018-02-20:
I'm currently having this issue with data that contains an extended source. Usually the background subtraction zone is selected an n-sigmas apart from the edge of the target extraction zone. What usually works for point sources does not work for extended sources Such as a [planetary nebula](http://simbad.u-strasbg.fr/simbad/sim-id?Ident=ic2448&NbIdent=1&Radius=2&Radius.unit=arcmin&submit=submit+id) (LEFT)
[image]
At least there should be a command line argument that allows to adjust the separation of the background zone.

simontorres, 2018-11-06:
Could this be a solution for #13 ?
Not for fully automatic.

simontorres, 2019-08-29:
related #298

simontorres, 2024-03-06:
Upon review with @Allonck we think that despite this issue being very old it is still a desired feature in 2024.