soar-telescope/goodman_pipeline #42: Modularization, Graphical Interfaces and Plug-ins

b1quint, 2017-03-15:
It seems that the users are looking for more graphical interactions. In that case, at some point we may need to decide if we want to implement it somehow. For now, I believe we should review how modular our package is. Based on that, I would like to leave this issue open for this kind of discussions. It was motivates by user's requirements like the one on Issue #41. 

I propose we build a block diagram with the modules that we have now and how they connect for the current package capabilities. This will help to visualize the current status of the project. Modules with bugs or erros should be properly flagged during this evaluation.

simontorres, 2017-03-15:
We talk about this in the past. And I think now is the right/best time to do it.

simontorres, 2018-07-25:
I think we moved away from this. I will close this issue now. Feel free to re-open if you think it is necessary.

closed 2018-07-25 by simontorres