soar-telescope/goodman_pipeline #47: Do we want to provide a spectrum normalization method?

simontorres, 2017-03-27:
People may want to have a normalized spectrum instead of a flux calibrated one. Do we want to provide that or let the user find another tool to do that?

b1quint, 2017-03-28:
How difficult it would be to implement this? And what would be the normalization factor? I believe that that could be left for the astronomers or for a future version.

simontorres, 2017-03-28:
Knowing the answer to your second question the implementation shouldn't be complicated. In other words, the technical part of the implementation doesn't seem to be a problem but it is important to know exactly what we are doing.

b1quint, 2017-03-28:
I would leave this as a low priority. Maybe for the release in the end of the year. Of course, that depends on how many users are interested in this feature. In my opinion, we should focus in the critical ones.

cbaorion, 2017-03-29:
Normalizing a spectrum is as simple as measuring counts at a given wavelength and dividing the spectrum by that number. Its always a question of to which specific wavelength or wavelength window you want to normalize to, but that should be input by the user. On the other hand, this is so easy and trivial that I wonder if its worth the trouble. This is something users can do on their wavelength calibrated, 1-D spectra, using IRAF or practically anything.
Let's not worry about it now.

simontorres, 2017-03-29:
so, this is also related to #42

simontorres, 2017-04-05:
I will close this issue now, we can reopen it later.

closed 2017-04-05 by simontorres