soar-telescope/goodman_pipeline #48: Check for different readout modes

simontorres, 2017-03-27:
# Problem
There might exist more than one readout mode in a night. Such data must not be mixed in any step of the process, currently the pipeline detects and warn (+ 10 second sleep) the user in case there are more than one modes but there is no control later on what should be combined with what.

# Solution (?).
The simplest solution is leave it as it is and maybe increase the "size of the warning"  this will depend on whether users do mix readout modes a lot or not.

In case they do I can add a check in every "image/spectrum" operation.

An alternative way would be to count the data for each readout mode and discard the smaller group, assuming the observer obtained that just for a test.

b1quint, 2017-03-29:
I believe that here we have some situations:

1) When the user takes calibration lamp after or before a target, (s)he changes the current ccd readout speed to 400 kHz so the readout is faster. In this case, this change should be ignored since the only important information is the position of the lines.

2) After that, the user may forget to change back the readout speed. In this case, there will be only one or more sequences of data with this. As I suggested in the issue #50, this decision is to the user to make. In this case we could:
2.1) Abort operations - stop everything.
2.2) Ignore differences - simply carry on ignoring the problem.
2.3) Select compatible - run the rest of the pipeline in the data where the readout speed is compatible with the ones obtained in the BIAS.

cbaorion, 2017-03-29:
I would leave the burden on the side of the user, to organize well his/her data before running the pipeline. For example, put different datasets in different folders.  Then, differences in readout should only be flagged as warnings during the reduction, that is, a warning message is issued when the pipeline encounters a different readout mode, but it still continues ahead with the processing.

simontorres, 2017-03-29:
I like what @cbaorion  said, although what @b1quint said is absolutely possible to do it might well take us a couple of weeks to implement and test. If we leave it like this we could focus on other matters. We could still do this for a later release.

b1quint, 2017-03-29:
For now, I would set the warnings and leave a further treatment as a TODO then.

simontorres, 2024-03-06:
Closing this issue because this is responsibility of the user as @cbaorion said.

closed 2024-03-06 by simontorres