soar-telescope/goodman_pipeline #50: No flat data/master flat found causes crash

simontorres, 2017-03-28:
# Description 
A master flat is created or read from existing ones there are some cases when neither is available and the program populates the master flat variable with a `None` object and in the next step when it tries to apply the flatfield correction the `None` will cause an AttributeError exception to be raised

# Proposed solutions
1. The simplest solution would be to check before if the master flat is not a None element. But that means there will be data that will not be flatfielded
2. Search for the _closest_ flat, i.e. the most similar flat in terms of some parameters to be defined.
3.  Skip the targets that is not possible to flatfield

# Note
- This is more complex in the spectroscopic mode.

b1quint, 2017-03-29:
I believe that this should raise a Warning to the user and, possibly, a question like: Do you want to continue without flat correction? If the user replies Yes, then simply carry on with the process without flat correction. If not, stop where you are and the user should start again.

cbaorion, 2017-03-29:
I think this issue needs different treatment in each of the imaging and the spectroscopic cases:
1) Imaging: flatfielding is necessary, not an option, so not having a flat field image should **pause** the processing and ask the user for input, expecting always a flatfield, but providing the option to specify a path to a different folder.
2) Spectroscopy: the user may not want to flatfield the data. In this case, there must exist the option of continuing **without** flatfielding.  The user should be prompted whether to continue or to use a flatfield from a different folder/directory.

simontorres, 2017-03-29:
There are still some improvements I want to add to this part in the code which should take us _a few steps further_ I think the suggestions are very useful.

There is an ERROR message printed, but I've found that requesting the user some _terminal interaction_  in the middle of a process supposed to be automatic doesn't feel right. I still can do it.

Regarding the use of a different folder or using calibrations from another directory was suggested by Jay at some point and I had them in my paper notes and now is issue #51

simontorres, 2017-04-03:
I have made some progress on this but still it doesn't solve the problem which is that it misses the correct identification.

The progress is a workaround that allows (only for "spectroscopic mode") the images to not be flatfielded and prints a warning.

And the problem is the following. The list of files should all fit in a "group" of images that will be processed together. But since OBSRA and OBSDEC are slightly different, maybe due to recentering the star into the slit or any other adjustment that changed the target RA and DEC values. This situation is very likely to happen quite often, I think, therefore it is necessary to develop a robust solution regarding this.

```
[------- 20161114_eng_3]$ hselect 02{29..36}* obsra obsdec grating slit

0229_comp_2100.fits     03:10:31.285    -68:35:59.798   SYZY_2100       5.00" long slit
0230_comp_2100.fits     03:10:31.285    -68:35:59.798   SYZY_2100       5.00" long slit
0231_quartz_2100.fits   03:10:31.285    -68:35:59.798   SYZY_2100       5.00" long slit
0232_quartz_2100.fits   03:10:31.285    -68:35:59.798   SYZY_2100       5.00" long slit
0233_quartz_2100.fits   03:10:31.285    -68:35:59.798   SYZY_2100       5.00" long slit
0234_EG21_2100.fits     03:10:31.580    -68:35:56.800   SYZY_2100       5.00" long slit
0235_EG21_2100.fits     03:10:31.580    -68:35:56.800   SYZY_2100       5.00" long slit
0236_EG21_2100.fits     03:10:31.580    -68:35:56.800   SYZY_2100       5.00" long slit
```

## Ideas for solution.
1.- One way to do it would be to create an argument such as `--critical-calibration` that tells the program that should work harder to find such groups of data (composed of COMP, FLAT and OBJECT). And if it still doesn't finds it just don't process the data. If we do the program to work harder every time we will have problem when we don't require critical calibrations to be performed.
2.- Find some other keyword that let us set apart the different groups of data.
3.- Use sky distance or time difference to associate data to groups.

simontorres, 2018-05-17:
Implemented error handling and prints the following message

```
[04:23:50PM][CRITICAL]: There is no FLAT images. Use --ignore-flats to continue without FLATs.
CRITICAL ERROR: FLAT not Found.
```
Using `--ignore-flats` process all data without flat fielding.

closed 2018-05-17 by simontorres