soar-telescope/goodman_pipeline #53: Allow other ways to store wavelength Solutions

simontorres, 2017-04-05:
This topic was brought up at the UNC meeting. The argument is that, since in order to write a linear solution in the header we are linearizing the data therefore we are re-sampling it. The request was to allow the data to be written as a table of wavelength and intensity (maybe multi-extension fits).

I need to look this up in more details. I would discard writting non-linear solutions in the header since the documentation is poor and is really complicated.

cbaorion, 2017-04-05:
I agree.

simontorres, 2019-09-04:
I'm still not in favor of writing fits files as tables because the model information is lost. 
- To recover it would be necessary to do a fit, again information is lost and errors not tracked.
- Some people might not need to recover the model information.

As an alternative solution for this is to use `adsf+gwcs`, I'm still not fully confidant with these tools but in general they allow to serialize arrays  and models.

For example, to write:

```pyhton
from asdf import AsdfFile

f = AsdfFile()

print(f.tree)
# will print {}

# adding attributes, poly is an astropy.modeling.Model instance, wavelength
# and intensity are arrays.
f.tree['model'] = poly
f.tree['wavelength'] = wavelength
f.tree['flux'] = intensity

# finally write to a file
f.write_to('testfile.asdf')
```

To read:
```python
ff = asdf.open('testfile.asdf')
# how access attributes or fields in the tree.
ff.tree['model'](range(len(intensity)))
```

simontorres, 2024-03-06:
We need to revisit this because of #384

simontorres, 2024-03-18:
This is kind of redundant, please focus attention in #384

closed 2024-03-18 by simontorres