soar-telescope/goodman_pipeline #54: Saturation: Remove saturated images

simontorres, 2017-04-05:
It is necessary to filter which images are to be removed. I would say only saturated flats have to be discarded.

cbaorion, 2017-04-05:
Yes, only saturated flats need to be removed

simontorres, 2017-04-27:
Done. It will act automatically and the default value is 55.000 counts. it can be changed using `--saturation`.

Example:
`--saturation 45000` will change the saturation level to 45000.

b1quint, 2017-05-29:
What is the algorithm for image rejection? I ran `redccd` and several flat images got entirely rejected because they contained a saturated line. This saturated line is pretty common and I believe a better criteria/handling should be performed.

b1quint, 2017-05-29:
Just reopening issue

simontorres, 2018-05-17:
We need to use `GAIN` to find the saturation level and then we have to define two criteria to check if a given pixel is saturated using electrons and ADU.

```python
SATURATION_ADU = 65536
SATURATION_E = 205000

def is_saturated(pixel_in_adu, gain):
    pixel_in_e = pixel_in_adu * gain
    if pixel_in_adu >= SATURATION_ADU or pixel_in_e >= SATURATION_E:
        return True
```

simontorres, 2018-10-11:
I'm very satisfied for the solution implemented in  PR  #245.

All the values of 50% full well were (re) tabulated and converted into a pandas DataFrame. So the saturation values are now exact or closer at least. 
It was defined that an image will be saturated if 1% or more of its pixels are above the saturation level, this can be changed using the command line argument ``--saturation``. The 1% default choice should be revised/approved by @cbaorion 

So, for some of the readout modes the saturation is reached before the 50% full well, for now I just ignored this on the basis that a decent scientist will not be doing science near saturation level. This can be discussed of course.

closed 2018-10-11 by simontorres