soar-telescope/goodman_pipeline #57: Update Trim Section

simontorres, 2017-04-26:
# Intro
The slit does not always cover the whole detector and this can create problems in the processes that follow, that's why we have decided to do a trimming according to the slit coverage on the detector. I've been doing some tests and detecting such limits is an easy task, what might be a problem is that those values are not constant and in principle you might think that it will vary with the slit but that is not True.

In fact is not clear at all. The plot bellow shows all the flat files taken during a single night and their respective "slit profile". 200 columns where averaged using median starting in the middle towards the red part of the spectrum.

[image]

cbaorion, 2017-04-26:
I don't quite understand the different plots. You will have to explain 
this graph to us in the Friday meeting.

Cesar

-- 

***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************

simontorres, 2017-04-27:
The simplest interpretation for the plot is that _The slit-illuminated region is not fixed and does not depend on the grating and slit alone_. So my solution was to implement the slit trimming section (commit 468463ccd82eac0b21ac46705a6c8945ee1cbb96) additional to the inital trimming. I left the initial trimming for completness. There are some cases when the slit trimming sections can't be found so for those cases the initial trimming should suffice.

I have tested it and so far it works fine.

closed 2017-04-27 by simontorres