soar-telescope/goodman_pipeline #58: The CCD reduction does not work if there is no Science Data.

simontorres, 2017-05-08:
Trying to process some engineering data taken during daytime the ccd pipeline did not process any data.
I have made some changes but so far can't get it to work and it appears to be a problem due to the time the data was taken, early in the morning, so it thinks it is in the previous night, most likely has to do with UTC midnight. 

I have some ideas and I will post here what solution I find.

simontorres, 2024-03-06:
@Allonck will test this and @simontorres will implement solutions.

Allonck, 2024-03-11:
I made a directory test without science images (only BIAS and FLATS), and here is the output:

```console
(goodman_pipeline) cmontalban@lswlan59 2021-11-18_test % redccd --debug --red-path RED_test_58
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[14:56:41][    INFO]: Starting Goodman HTS Pipeline Log [core.setup_logging:2938]
[14:56:41][    INFO]: Local Time    : 2024-03-11 14:56:41.204070 [core.setup_logging:2939]
[14:56:41][    INFO]: Universal Time: 2024-03-11 17:56:41.204119 [core.setup_logging:2941]
[14:56:41][   DEBUG]: Starting new HTTPS connection (1): api.github.com:443 [connectionpool._new_conn:1055]
[14:56:41][   DEBUG]: https://api.github.com:443 "GET /repos/soar-telescope/goodman/releases/latest HTTP/1.1" 301 213 [connectionpool._make_request:549]
[14:56:42][   DEBUG]: https://api.github.com:443 "GET /repositories/62915938/releases/latest HTTP/1.1" 200 847 [connectionpool._make_request:549]
[14:56:42][   DEBUG]: Starting new HTTPS connection (1): api.github.com:443 [connectionpool._new_conn:1055]
[14:56:42][   DEBUG]: https://api.github.com:443 "GET /repos/soar-telescope/goodman/releases/latest HTTP/1.1" 301 213 [connectionpool._make_request:549]
[14:56:42][   DEBUG]: https://api.github.com:443 "GET /repositories/62915938/releases/latest HTTP/1.1" 200 847 [connectionpool._make_request:549]
[14:56:42][    INFO]: Pipeline Version: 1.3.7 (latest) [core.setup_logging:2952]
[14:56:42][   DEBUG]: Initializing DataClassifier instance [goodman_ccd.__init__:168]
[14:56:42][   DEBUG]: Starting argument checks. [goodman_ccd._check_args:274]
[14:56:42][   DEBUG]: Raw data folder "/Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test" exists. [goodman_ccd._check_args:278]
[14:56:42][    INFO]: Found 51 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test. [goodman_ccd._check_args:289]
[14:56:42][ WARNING]: Reduction folder doesn't exist. [goodman_ccd._check_args:359]
[14:56:42][    INFO]: Created directory for reduced data. [goodman_ccd._check_args:361]
[14:56:42][    INFO]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test/RED_test_58 [goodman_ccd._check_args:362]
[14:56:42][   DEBUG]: Calling data_classifier Instance of DataClassifier [goodman_ccd.__call__:206]
[14:56:42][   DEBUG]: keywords in setter before pruning: *. [image_collection.keywords:231]
[14:56:42][   DEBUG]: keywords after pruning ['*']. [image_collection.keywords:239]
[14:56:42][   DEBUG]: should be building new table... [image_collection.keywords:253]
[14:56:42][   DEBUG]: Raw path: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test [data_classifier.__call__:77]
[14:56:42][   DEBUG]: Detected Red camera. [data_classifier._get_instrument:124]
[14:56:42][    INFO]: Instrument: Red Camera [data_classifier.__call__:81]
[14:56:42][    INFO]: Detected Spectroscopy Data from Red Camera [data_classifier._get_obs_technique:159]
[14:56:42][    INFO]: Observing Technique: Spectroscopy [data_classifier.__call__:90]
[14:56:42][   DEBUG]: Initializing night organizer procedure [goodman_ccd.__call__:237]
[14:56:42][   DEBUG]: Calling night organizer procedure [goodman_ccd.__call__:245]
[14:56:42][   DEBUG]: keywords in setter before pruning: ['naxis', 'date', 'slit', 'date-obs', 'obstype', 'object', 'exptime', 'obsra', 'obsdec', 'grating', 'cam_targ', 'grt_targ', 'filter', 'filter2', 'gain', 'rdnoise', 'roi', 'wavmode']. [image_collection.keywords:231]
[14:56:42][   DEBUG]: keywords after pruning ['rdnoise', 'object', 'gain', 'date-obs', 'obsra', 'obstype', 'cam_targ', 'filter2', 'naxis', 'exptime', 'grating', 'filter', 'obsdec', 'wavmode', 'grt_targ', 'roi', 'slit', 'date']. [image_collection.keywords:239]
[14:56:42][   DEBUG]: should be building new table... [image_collection.keywords:253]
[14:56:43][    INFO]: Organizing data for this configuration: Gain: 1.48, Noise: 3.89, ROI: Spectroscopic 2x2 [night_organizer.__call__:140]
[14:56:53][   DEBUG]: Sun Set 2021-11-18T23:15:44.494 [core.get_twilight_time:1908]
[14:56:53][   DEBUG]: Sun Rise 2021-11-18T09:40:57.560 [core.get_twilight_time:1909]
[14:56:53][    INFO]: Found valid data, appending to data container list [night_organizer.__call__:178]
[14:56:53][   DEBUG]: Initializing image processing procedure [goodman_ccd.__call__:259]
[14:56:53][   DEBUG]: Calling image processing procedure. [goodman_ccd.__call__:264]
[14:56:53][ WARNING]: Determining trim section. Assuming you have only one kind of data in this folder [core.define_trim_section:1039]
[14:56:53][    INFO]: Trim Section: [26:2055,2:948] [core.define_trim_section:1077]
[14:56:53][   DEBUG]: Overscan Sample File /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test/0071_BIAS_18-11-2021.fits [core.get_overscan_region:1611]
[14:56:53][    INFO]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3 [core.get_overscan_region:1625]
[14:56:53][    INFO]: Overscan Region: [3:24,1:948] [core.get_overscan_region:1677]
[14:56:53][   DEBUG]: Creating Master Bias [image_processor.__call__:105]
[14:56:53][    INFO]: Creating master bias [core.create_master_bias:400]
[... cut ...]
[14:57:07][    INFO]: Created Master Flat: RED_test_58/master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits [core.create_master_flats:563]
(goodman_pipeline) cmontalban@lswlan59 2021-11-18_test % 

```

The red-path directory only has:
* master_bias_RED_SP_2x2_R03.89_G01.48.fits
* master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
* master_flat_400m2_GG455_1.0_dome.fits

What I see is just the creation of masters. In fact, the flats were not normalized, and when comparing the counts with the RAW images, I would say that they were not bias subtracted either. In summary, the images weren't processed.

Allonck, 2024-03-11:
I made another test with the ``Blue`` camera and I have the same results.

```console 
(goodman_pipeline) cmontalban@lswlan59 2023-03-09_test_58 % redccd --debug --red-path RED_test_58
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[15:12:22][    INFO]: Starting Goodman HTS Pipeline Log [core.setup_logging:2938]
[15:12:22][    INFO]: Local Time    : 2024-03-11 15:12:22.163553 [core.setup_logging:2939]
[15:12:22][    INFO]: Universal Time: 2024-03-11 18:12:22.163601 [core.setup_logging:2941]
[15:12:22][   DEBUG]: Starting new HTTPS connection (1): api.github.com:443 [connectionpool._new_conn:1055]
[15:12:23][   DEBUG]: https://api.github.com:443 "GET /repos/soar-telescope/goodman/releases/latest HTTP/1.1" 301 213 [connectionpool._make_request:549]
[15:12:23][   DEBUG]: https://api.github.com:443 "GET /repositories/62915938/releases/latest HTTP/1.1" 200 847 [connectionpool._make_request:549]
[15:12:23][   DEBUG]: Starting new HTTPS connection (1): api.github.com:443 [connectionpool._new_conn:1055]
[15:12:24][   DEBUG]: https://api.github.com:443 "GET /repos/soar-telescope/goodman/releases/latest HTTP/1.1" 301 213 [connectionpool._make_request:549]
[15:12:24][   DEBUG]: https://api.github.com:443 "GET /repositories/62915938/releases/latest HTTP/1.1" 200 847 [connectionpool._make_request:549]
[15:12:24][    INFO]: Pipeline Version: 1.3.7 (latest) [core.setup_logging:2952]
[15:12:24][   DEBUG]: Initializing DataClassifier instance [goodman_ccd.__init__:168]
[15:12:24][   DEBUG]: Starting argument checks. [goodman_ccd._check_args:274]
[15:12:24][   DEBUG]: Raw data folder "/Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58" exists. [goodman_ccd._check_args:278]
[15:12:24][    INFO]: Found 37 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58. [goodman_ccd._check_args:289]
[15:12:24][ WARNING]: Reduction folder doesn't exist. [goodman_ccd._check_args:359]
[15:12:24][    INFO]: Created directory for reduced data. [goodman_ccd._check_args:361]
[15:12:24][    INFO]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58/RED_test_58 [goodman_ccd._check_args:362]
[15:12:24][   DEBUG]: Calling data_classifier Instance of DataClassifier [goodman_ccd.__call__:206]
[15:12:24][   DEBUG]: keywords in setter before pruning: *. [image_collection.keywords:231]
[15:12:24][   DEBUG]: keywords after pruning ['*']. [image_collection.keywords:239]
[15:12:24][   DEBUG]: should be building new table... [image_collection.keywords:253]
[15:12:24][ WARNING]: unable to get FITS header for file /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58/repeated_fits: [Errno 21] Is a directory: '/Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58/repeated_fits'. [image_collection._fits_summary:623]
[15:12:24][   DEBUG]: Raw path: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58 [data_classifier.__call__:77]
[15:12:24][   DEBUG]: Detected Blue camera. [data_classifier._get_instrument:124]
[15:12:24][    INFO]: Instrument: Blue Camera [data_classifier.__call__:81]
[15:12:24][    INFO]: Detected Spectroscopy Data from Blue Camera [data_classifier._get_obs_technique:159]
[15:12:24][    INFO]: Observing Technique: Spectroscopy [data_classifier.__call__:90]
[15:12:24][   DEBUG]: Initializing night organizer procedure [goodman_ccd.__call__:237]
[15:12:24][   DEBUG]: Calling night organizer procedure [goodman_ccd.__call__:245]
[15:12:24][   DEBUG]: keywords in setter before pruning: ['naxis', 'date', 'slit', 'date-obs', 'obstype', 'object', 'exptime', 'obsra', 'obsdec', 'grating', 'cam_targ', 'grt_targ', 'filter', 'filter2', 'gain', 'rdnoise', 'roi', 'wavmode']. [image_collection.keywords:231]
[15:12:24][   DEBUG]: keywords after pruning ['filter2', 'obsdec', 'object', 'cam_targ', 'grt_targ', 'naxis', 'exptime', 'roi', 'slit', 'obsra', 'wavmode', 'rdnoise', 'filter', 'date-obs', 'obstype', 'date', 'gain', 'grating']. [image_collection.keywords:239]
[15:12:24][   DEBUG]: should be building new table... [image_collection.keywords:253]
[15:12:24][ WARNING]: unable to get FITS header for file /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58/repeated_fits: [Errno 21] Is a directory: '/Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_test_58/repeated_fits'. [image_collection._fits_summary:623]
[15:12:24][    INFO]: Organizing data for this configuration: Gain: 1.40, Noise: 4.74, ROI: Spectroscopic 2x2 [night_organizer.__call__:140]
[15:12:35][   DEBUG]: Sun Set 2023-03-09T23:03:26.364 [core.get_twilight_time:1908]
[15:12:35][   DEBUG]: Sun Rise 2023-03-09T10:42:57.770 [core.get_twilight_time:1909]
[15:12:35][    INFO]: Found valid data, appending to data container list [night_organizer.__call__:178]
[15:12:35][   DEBUG]: Initializing image processing procedure [goodman_ccd.__call__:259]
[15:12:35][   DEBUG]: Calling image processing procedure. [goodman_ccd.__call__:264]
[15:12:35][ WARNING]: Determining trim section. Assuming you have only one kind of data in this folder [core.define_trim_section:1039]
[15:12:35][    INFO]: Trim Section: [26:2055,2:948] [core.define_trim_section:1077]
[... cut ...]
[15:12:41][    INFO]: Combining 15 images to create master flat [core.create_master_flats:544]
[15:12:45][    INFO]: Saving FITS file to master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits [core.write_fits:3323]
[15:12:45][    INFO]: Created Master Flat: RED_test_58/master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits [core.create_master_flats:563]
(goodman_pipeline) cmontalban@lswlan59 2023-03-09_test_58 % 
```

The masters created:

* master_bias_BLUE_SP_2x2_R04.74_G01.40.fits
* master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits

Allonck, 2024-03-11:
I made a test with only BIAS imaging data and the ``redccd`` process not even started.

```terminal

(goodman_pipeline) cmontalban@lswlan59 imaging % redccd 
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[15:27:20][I]: Starting Goodman HTS Pipeline Log
[15:27:20][I]: Local Time    : 2024-03-11 15:27:20.103691
[15:27:20][I]: Universal Time: 2024-03-11 18:27:20.103739
[15:27:22][I]: Pipeline Version: 1.3.7 (latest)
[15:27:22][I]: Found 21 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_GP/imaging.
[15:27:22][I]: No special reduced data path defined. Proceeding with defaults.
[15:27:22][W]: Reduction folder doesn't exist.
[15:27:22][I]: Created directory for reduced data.
[15:27:22][I]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2023-03-09/2023-03-09_GP/imaging/RED
[15:27:22][C]: Unable to determine which camera was used.
[15:27:22][I]: Make sure you only have 'Blue' or 'Red' camera data only, not both.
```
And the output in the ``RED`` directory is empty.

Here is the header of one BIAS image:

>TELESCOP= 'SOAR 4.1m'                                                           
INSTRUME= 'ghts_blue_imager'                                                    
OBJECT  = 'BIAS_200ATTN0_IM_2x2'                                                
NOTES   = ''                                                                    
INSTCONF= 'Blue'                                                                
CAMSWV  = 'Beta 29'                                                             
CONSWV  = '8359,10'                                                             
INSTSWV = 'GSCS 2.0.17.6.6-Alpha'                                               
CAMERA  = 'Si 600-658 '                                                         
RA      = '08:24:36.242'       / right ascension [hh:mm:ss.sss]                 
DEC     = '-17:33:38.374'      / declination [dd:mm:ss.sss]                     
AIRMASS =             1.660000 / airmass at approx. start of exposure           
UT      = '22:13:10.480'       / time at approx. start of exposure [UTC]        
FOCUS   =          -894.210000 / SOAR telescope focus                           
MOUNT_AZ=            90.007236 / SOAR mount azimuth                             
MOUNT_EL=            37.012836 / SOAR mount elevation                           
ROTATOR =            90.000000 / Nasymth cage rotator angle [deg]               
POSANGLE=             0.000000 / position angle [deg. E of N]                   
SEEING  =            -1.000000 / DIMM seeing [arcsec]                           
LST     = '04:39:36.550'       / Local Sidereal Time  [hh:mm:ss.sss]            
OBSRA   = '18:09:43.711'       / target right ascension [hh:mm:ss.sss]          
OBSDEC  = '-2:19:47.500'       / target declination  [hh:mm:ss.sss]             
DOME_AZ =           180.006404 / SOAR Dome Shutter azimuth                      
HA      = '-03:45:00.695'      / Hour Angle  [hh:mm:ss.sss]                     
ADCSTAT = 'IN'                 / ADC Status                                     
ADCPOS  = 'DONE_50.000'        / ADC Position                                   
LAMP_HGA= 'FALSE'              / Hg(Ar)                                         
LAMP_NE = 'FALSE'              / Neon                                           
LAMP_AR = 'FALSE'              / Argon                                          
LAMP_FE = 'FALSE'              / Iron                                           
LAMP_CU = 'FALSE'              / Copper                                         
LAMP_QUA= 'FALSE'              / Quartz                                         
LAMP_QPE=             0.000000 / Quartz Percent                                 
LAMP_BUL= 'FALSE'              / Bulb                                           
LAMP_DOM= 'FALSE'              / Dome                                           
LAMP_DPE=             0.000000 / Dome Percent                                   
CAM_ANG =             0.000111 / camera angle [deg]                             
GRT_ANG =             0.661300 / grating angle [deg]                            
CAM_TARG=             0.000000 / camera target [deg]                            
GRT_TARG=             0.661000 / grating target [deg]                           
CAM_FOC =                  402 / camera focus                                   
COLL_FOC=                 1001 / collimator focus                               
FILTER  = 'NO_FILTER'          / primary filter wheel                           
FILTER2 = 'NO_FILTER'          / secondary filter wheel                         
GRATING = 'NO_GRATING'         / VPH grating [1/mm]                             
SLIT    = 'NO_MASK'            / slit [arcsec]                                  
COL_TEMP=            14.296875 / coll ext temp (deg C)                          
[... 10 lines cut ...]
ENVWIN  =             5.104080 / Wind Speed [km/hr] at start of exposure        
ENVPRE  =           737.750000 / Atmospheric Pressure [hPS] at start of exposur 
ENVDIR  =           136.100000 / Wind Direction at start of exposure            
ENVTEM  =            15.000000 / Outside Temperature [C] at start of exposure   
ENVHUM  =            63.500000 / Relative Humidity at start of exposure         
[... cut ...]
BLKUID  =                    0                                                  
L1PUBDAT= '2023-03-09T22:13:12.153296Z'                                         
IMAGEID = '77'                                                                  
HEADERVE= '1.0.1'                                                               
BLOCK_ID=     1678399931761036                                                  
COMMENT  Triggered, Exp Time= 00 - Image Transform: Flip Y - Image Transform: Fl
COMMENT  ip Y, Saved as: 0077_BIAS_09-03-2023.fits                              
END

Allonck, 2024-03-11:
It occurs to me that the problem happens because the whole process revolves around science images. If the process happens the same as in the Jupyter notebook (Create and bias-correct science images, create masterflat [masterflats are bias-corrected when created] and then normalize it for science image correction), then it makes sense that calibration images are not processed.

Changing this I understand may be complex and I don't know to what extent advantageous, perhaps to better structure the software so that it can at least process calibration images?

If instead, all images are processed by BIAS, then the masterFLAT is created with these images processed by BIAS and everything is corrected by masterFLAT (except the BIAS) [After being normalized] could be a solution to this problem.

Allonck, 2024-03-22:
Fixed in #395

closed 2024-03-22 by Allonck