soar-telescope/goodman_pipeline #62: Reduced data is useless for Blue Camera (SOLVED)

simontorres, 2017-05-13:
Yesterday I found a very important bug. The image would have an horrible pattern which made the data useless. The most important part was that it had negative values where it shouldnt. 

In the following (debug) image the Blue line is the combined master flat. The black line is the model, never mind the fact that it looks a really bad fit, it is, I was experimenting with different orders when I captured this image. But the result was always the same because the source of the problem was elsewhere. And the Red line is the _normalized_ flat.
[image]

It is very clear that something terrible wrong is going on here. I was able to trace the problem to the overscan subtraction part of the process. The overscan regions are different. I didn't check if this was the case before the change of Blue computer on Goodman or just appeared now.

```python
# Assuming ROI: Spectroscopic 1x1
blue_overscan_region = '[1:16,1:4096]'
red_overscan_region = '[5:45,1:4096]'
```
There is some work to be done here, since this will change for different ROI.

simontorres, 2017-05-13:
I have tested the pipeline with the new overscan for the Blue Camera and it works OK.

simontorres, 2017-05-15:
I solved this at least for Spectroscopic 1x1, 2x2 and 3x3.

```python
ccd = CCDData.read(sample_image, unit=u.adu)
serial_binning, paralell_bining = [int(x) for x in ccd.header['CCDSUM'].split()]
log.info('Using predefined overscan region for binning 1x1')
if self.technique == 'Spectroscopy':
    if self.instrument == 'Red':
        overscan_region = '[{:d}:{:d},1:1896]'.format(int(np.ceil(6. / serial_binning)), int(49./ serial_binning))
    elif self.instrument == 'Blue':
        overscan_region = '[1:{:d},1:1896]'.format(int(16. / serial_binning))
```

At this moment the fixed `1896` value will not create any problem unless we want to use a different value. In other words for binning 2x2 and 3x3 will be smaller but being a larger number doesn't create a problem.

b1quint, 2017-05-26:
If this is solved, I belive we can close this issue, no?

simontorres, 2017-05-26:
It is possible. I think I didn't close it because the spatial axis is fixed (last paragraph in the solution post). I you think it is not necessary to change that we can close the issue.

simontorres, 2017-05-29:
Decision: use the following way:

```python
 h = ccd.data.shape[0]  # Image height - spatial direction
 w = ccd.data.shape[1]  # Image width - spectral direction

 # Take the binnings
 serial_binning, parallel_binning = [int(x) for x in ccd.header['CCDSUM'].
 if self.technique == 'Spectroscopy':
    log.info('Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3')
    if self.instrument == 'Red':
        # for red camera it is necessary to eliminate the first rows/columns (depends on the point of
        # view) because they come with an abnormal high signal. Usually the first 5 pixels.
        # In order to find the corresponding value for the subsequent binnings divide by the binning
        # size.
        # The numbers 6 and 49 where obtained from visual inspection
        overscan_region = '[{:d}:{:d},1:{:d}]'.format(int(np.ceil(6. / serial_binning)), int(49. / serial_binning), w)
    elif self.instrument == 'Blue':
        # 16 is the length of the overscan region with no binning.
        overscan_region = '[1:{:d},1:{:d}]'.format(int(16. / serial_binning), w)

```

simontorres, 2018-07-12:
Solved

closed 2018-07-12 by simontorres