soar-telescope/goodman_pipeline #69: Cleaner output

b1quint, 2017-05-29:
We need a clearer output. At the moment, the output contains too much and unnecessary information. Here is a suggestion of format class:

    class MyLogFormatter(logging.Formatter):

        # err_fmt = "%(msg)s"
        dbg_fmt = "[D] %(module)s: %(lineno)d: %(msg)s"
        info_fmt = " %(msg)s"
        # warn_fmt = "W %(msg)s"

    def __init__(self, fmt="%(levelno)s: %(msg)s"):
        logging.Formatter.__init__(self, fmt)

    def format(self, record):

        # Save the original format configured by the user
        # when the logger formatter was instantiated
        format_orig = self._fmt

        # Replace the original format with one customized by logging level
        if record.levelno == logging.DEBUG:
            self._fmt = MyLogFormatter.dbg_fmt

        elif record.levelno == logging.INFO:
            self._fmt = MyLogFormatter.info_fmt

        # elif record.levelno == logging.ERROR:
        #    self._fmt = MyLogFormatter.err_fmt

        # elif record.levelno == logging.WARNING:
        #    self._fmt = MyLogFormatter.warn_fmt

        # Call the original formatter class to do the grunt work
        result = logging.Formatter.format(self, record)

        # Restore the original format configured by the user
        self._fmt = format_orig

        return result

If correctly implemented this should handle the issue but I still could not manage properly logs and the Python logging system to effectively suggest how to do it.

b1quint, 2017-05-29:
INFO logging should be much cleaner but the DEBUG mode should contain all the information.

Also, the DEBUG mode have to display what is the current image that is being modified to help to track issues.

simontorres, 2017-05-29:
This is done for the methods/functions that remove conflictive and duplicated keywords and for methods that import flat and bias images prior to combination

simontorres, 2018-02-20:
I think I didn't pay attention when the issue was created. I did an experiment with a different way of using `logging` the problem was that there is little control over the desired logging level. Most likely I will end up implemented that one which relies in a configuration file. The problem with the current way (or the in-code configuration for `logging`) is that not all logging messages get _"emitted"_ when they should.
Also in the current latest version the format for the logger is simpler when the level is set to `INFO` than when it is set to `DEBUG`.

```python
if '--debug' in sys.argv:
    FORMAT = '%(levelname)s: %(asctime)s:%(module)s.%(funcName)s: %(message)s'
else:
    FORMAT = '%(levelname)s: %(asctime)s: %(message)s'
DATE_FORMAT = '%I:%M:%S%p'
LOG_FILENAME = 'goodman_spec.log'
...

...
# somewhere else in the code
if args.log_file != LOG_FILENAME:
    LOG_FILENAME = args.log_file

log.info('Logging to file {:s}'.format(LOG_FILENAME))
file_handler = logging.FileHandler(LOG_FILENAME)
formatter = logging.Formatter(fmt=FORMAT, datefmt=DATE_FORMAT)
file_handler.setFormatter(fmt=formatter)
log.addHandler(file_handler)

if args.debug_mode:
    log.info('Changing log level to DEBUG.')
    log.setLevel(level=logging.DEBUG)

```
I think right now is not a priority but surely we will have to address this.

simontorres, 2018-07-12:
I think we already solved this with #181

closed 2018-07-12 by simontorres