soar-telescope/goodman_pipeline #72: End of night calibrations

b1quint, 2017-05-29:
Calibration data obtained at the end of the night causes science files to be ignored.

b1quint, 2017-05-29:
One possible solution is to implement a criterium of "belong to the night?" This will be true if the file was obtained within the range of +/- 12h counting from the midnight (example reference 2017-05-29 23:59 CLT).

simontorres, 2017-05-29:
We decided to process everything that is inside the folder. The only calibration that needs to time-checked are flats and comparison lamps. The criteria would be position in the sky (`OBSRA` and `OBSDEC`). If no calibration is found near to the targets get the any in the folder, provided it is valid.

simontorres, 2017-08-16:
This issue has been solved but there is a side effect described in Issue #101

closed 2017-08-16 by simontorres