soar-telescope/goodman_pipeline #76: Trim Section is not defined dynamically

simontorres, 2017-05-29:
The trim section is defined in a static way. It must be done dynamically depending on the binning and ROI.

simontorres, 2017-05-30:
In fact is so static that completely forgot Imaging. Solving this today.

simontorres, 2017-05-30:
Fixed this, for imaging it takes `TRIMSEC` from header.
Some testing is needed.

simontorres, 2018-05-17:
What happens to custom ROI?
 - Should we included a table of trim section values?

cbaorion, 2018-05-18:
Let's is not worry about custom ROIs.  If somebody wants to use the pipeline for that, let them modify it themselves.  The documentation within the pipeline should allow users that want to go that way, to modify it to suit their very specific needs. Otherwise you will spend the rest of your life trying to make everybody coming with any possible custom ROI happy.  That is not the goal of the pipeline.

simontorres, 2018-05-22:
Then, rather than finding the values ourselves we should be adding built-in flexibility for instance with #102

simontorres, 2020-03-09:
There are no current issues with TRIM sections

closed 2020-03-09 by simontorres