soar-telescope/goodman_pipeline #77: Allow to use either filter wheel for either observing technique

simontorres, 2017-05-30:
Right now the filter wheel 1 `FILTER` is dedicated to imaging filters and filter wheel 2 `FILTER2` is dedicated to order blocking filters. 

It would be good to flexibilize this.

simontorres, 2018-02-20:
I think we should keep it as it is right now. It would complicate things even more.

closed 2018-02-20 by simontorres

cbaorion, 2018-02-20:
I agree.

Cesar

-- 
********************************************
* César Briceño                            *
* ******************************************
* Cerro Tololo Inter-American Observatory  *                                     *
* Casilla 603                              *
* La Serena 1700000, CHILE                 *
* Tlf: +56-51-2205294                      *
* Email: cbriceno@ctio.noao.edu            *
********************************************