soar-telescope/goodman_pipeline #81: Migration towards new method of Identify and extract spectra

simontorres, 2017-06-05:
I have partially implemented a new method for doing optimal extraction, although the optimal extraction is not complete yet, the identification and extraction methods work much better than the old ones, so I will include them in the beta version, I will have to do some big changes but the final product will be better.

simontorres, 2017-06-23:
The migration has been completed. This eliminated the need of the module `goodman_spec.process.py` and now all the functions are in `goodman_ccd.core.py` they can be imported from any spectroscopic module as follows:

```python
# example of how to import a function from core.py
from goodman_ccd.core import spectroscopic_extraction
```

closed 2017-06-23 by simontorres

cbaorion, 2017-06-23:
Acknowledged. Great.