soar-telescope/goodman_pipeline #85: Repair cosmic Ray Rejection implementation

simontorres, 2017-06-15:
Add ability to use the cosmic ray rejection mask created in the first (ccd) pipeline
* when trimming the image `ccd.data`, the `ccd.mask` attribute must be trimmed as well.

If there hasn't been a cosmic ray rejection the `mask` attribute is a `None`

simontorres, 2017-06-20:
The cosmic ray rejection mechanism is not working properly. We are using astropy' LACosmic, David explained to me that LACosmic was not designed for spectroscopy and suggested to use [this code](http://users.camk.edu.pl/pych/DCR/). I installed it and tried and seems to work much better. See the animation in full screen and zoom it if necessary.

[image]

simontorres, 2017-06-21:
I have finally implemented the `dcr` code in the redccd. There are some small issues to be solved like the `dcr.par` file required by `dcr` this is hard coded and can't be changed. The user will have to place a modified version in the raw_data directory or the program will copy a default one.

simontorres, 2017-06-23:
It works well. In general this will require the user to compile a code. For what I've read it is possible to compile C code when running `python setup.py install` by configuring `setup.py` accordingly but I think this should be left for version 1.0.

cbaorion, 2017-06-23:
This is ok for the beta version.

simontorres, 2017-08-23:
This has been fully implemented. There is a possible problem though for data with binning different than 1x1 in such case it is necessary to change the parameters for dcr at `dcr.par`

## Default dcr.par

```shell
THRESH	= 3.0	// Threshold (in STDDEV)
XRAD	= 9	// x-radius of the box (size = 2 * radius)
YRAD	= 9	// y-radius of the box (size = 2 * radius)
NPASS	= 5	// Maximum number of cleaning passes
DIAXIS	= 0	// Dispersion axis: 0 - no dispersion, 1 - X, 2 - Y
LRAD	= 1	// Lower radius of region for replacement statistics
URAD	= 3	// Upper radius of region for replacement statistics
GRAD	= 1	// Growing radius
VERBOSE	= 1	// Verbose level [0,1,2]
END
```

## Modified dcr.par for data with binning 2x2

```shell
THRESH  = 2.0   // Threshold (in STDDEV)
XRAD    = 9     // x-radius of the box (size = 2 * radius)
YRAD    = 9     // y-radius of the box (size = 2 * radius)
NPASS   = 5     // Maximum number of cleaning passes
DIAXIS  = 0     // Dispersion axis: 0 - no dispersion, 1 - X, 2 - Y
LRAD    = 1     // Lower radius of region for replacement statistics
URAD    = 3     // Upper radius of region for replacement statistics
GRAD    = 0     // Growing radius
VERBOSE = 1     // Verbose level [0,1,2]
END
```

You can find the description of those parameter in the paper. http://adsabs.harvard.edu/abs/2004PASP..116..148P

simontorres, 2017-09-27:
The discussion of parameters variation is discussed here #105

closed 2017-09-27 by simontorres