soar-telescope/goodman_pipeline #86: [SOLVED] Problem finding binary using PyCharm on Ubuntu

simontorres, 2017-06-23:
When I was testing the new implementation for the [new cosmic ray rejection tool](http://users.camk.edu.pl/pych/DCR/) I kept getting errors where the command interpreter wasn't able to find the `ecr` binary or could not read properly the `ecr.par` file. All this when running the program from PyCharm,

The simplest solution was to, instead of starting PyCharm from Ubuntu Launcher do it from a terminal.

There is a second posible solution but so far didn't work for me and it is to add the path to pycharm.

cbaorion, 2017-06-23:
I would in general avoid running programs from specialized plataforms 
like PyCharm. Let's just make sure the pipeline can be run from the 
terminal/console, in the good-ol fashioned way.  We can get fancy later, 
maybe for version V1.

-- 

***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************

simontorres, 2017-06-23:
Absolutely. But we need to be able to run it directly from pycharm for development purposes, otherwise the enormous benefits from using it will be canceled out by this.
This is not for the users but this issue is useful for documentation purposes, even for other people searching in google.

cbaorion, 2017-06-23:
Ok. Agreed.

-- 

***********************************************
* César Briceño                               *
* *********************************************
* AURA/CTIO                                   *
* Observatorio Interamericano de Cerro Tololo *
* Casilla 603                                 *
* La Serena, Chile                            *
* Email: cbriceno@ctio.noao.edu               *
*        cbaorion2008@gmail.com               *
***********************************************

simontorres, 2017-08-16:
This issue is actually solved.

closed 2017-08-16 by simontorres