soar-telescope/goodman_pipeline #91: gp-pages is like an independent repository

b1quint, 2017-06-30:
According to the GitHub and GitPages Documentation, the `gh-pages` branch in a project can be defined as a completely different project where we can work only in the online documentation and how it is shown to the user. 

For now, we need to isolate what is being developed from what will be shown in the page in different branches. With this, the user will be able to  access the project's page by the following address:

https://soar-telescope.github.io/goodman 

Which is much more elegant than

https://github.com/soar-telescope/goodman

The first one is the one that should be used when distributing the package.

simontorres, 2017-06-30:
This is how you can access the GitPages Documentation, the missing part is that is a branch within the repository and not an independent repository.

simontorres, 2017-08-23:
I need help setting this up. @b1quint 

The sphinx-generated html pages are stored in a directory called `html` in the root folder of the branch `gh-pages` There are many files some of them are mind. @b1quint Could you delete all and put the functional template in there? I can take care of putting the html docstrins back in there.
Let me know.

simontorres, 2018-05-17:
we have moved documentation to readthedocs goodman.readthedocs.org/en/latest

closed 2018-05-17 by simontorres