soar-telescope/goodman_pipeline #102: Using a configuration file

simontorres, 2017-08-21:
When trying to quickly reduce last week's TOO I realized that the potential of the pipeline, that we initially intended, is there, in other words, the functions or methods are capable of doing more than what is currently being requested by the _intelligence_ implemented on the pipeline, (this is in general). 

There are several ways to exploit these capabilities.

- Add more intelligence to the pipeline which is time consuming and should be left to the end (I agree  with @b1quint on this one)
- Add more command line arguments in which case we could end up having potentially hundred of them, which again is not desired 
- The last option is to use a configuration file. I initially though using a `conf.py` file but after reading [this post](https://martin-thoma.com/configuration-files-in-python/) I'm more inclined to using _json_.

Finally, if we successfully evolve the command line pipeline into a GUI version this would work very good because it would let us modify the parameters from the GUI itself. I'm sure there are many possible ways to do this, so I would appreciate some feedback.

b1quint, 2017-08-21:
I believe that having a .json or a .xml configuration for each run could be very productive and I would like to support that decision. 

That would make it easier to implement future GUI(s). In any case, that would require our software to handle these config files and to have default parameters well implemented. I believe that python and other languages have wrappers to deal with JSON/XML files so I believe that it would be a good strategy.

cbaorion, 2017-08-21:
I agree. I would go for the JSON format, since its much simpler than XLM with all of the markup tags, hence easier to use and create/edit with a plain text editor like VI,Emacs, NotePad or whatever flavour one prefers. Its already in use to setup observations for DECam and the Mayall MzLS Survey, which means an increasingly larger community of users is already getting used to JSON files.

simontorres, 2017-08-23:
I'm reopening this issue because we need it for feedback.

b1quint, 2017-09-04:
For now, let's have it defined the JSON as our future standard for configuration files. 

The issue shall remain open until the use of configuration files is implemented.

b1quint, 2018-05-04:
#129

b1quint, 2018-05-04:
We have to define a format and just stick with it. When it works, we can try other formats. JSON is easy for users to create/edit.

cbaorion, 2018-05-04:
Again, let's just use JSON.  Unless somebody convinces me of the need for something else I vote for JSON.

simontorres, 2020-12-01:
I think we've got really far without a configuration file, adding one now would just disrupt the development process too much.

closed 2020-12-01 by simontorres