soar-telescope/goodman_pipeline #107: List of Instrument Configurations Supported by the pipeline

simontorres, 2017-09-27:
The currently supported instrument configuration supported are listed in the user manual but there should be a list of supported modes that we will support or aim to support.

simontorres, 2018-02-20:
We are going to support all spectroscopic modes for gratings `400`, `600`, `930` and `1200`.

closed 2018-02-20 by simontorres

cbaorion, 2018-02-20:
Great.  Does it mean its already implemented?

Cesar

-- 
********************************************
* César Briceño                            *
* ******************************************
* Cerro Tololo Inter-American Observatory  *                                     *
* Casilla 603                              *
* La Serena 1700000, CHILE                 *
* Tlf: +56-51-2205294                      *
* Email: cbriceno@ctio.noao.edu            *
********************************************

simontorres, 2018-02-20:
As long as we complete this large list of lamps https://docs.google.com/spreadsheets/d/1uVCtgdfWLt-PNebhwPKUEvPVwYoPMEJjFWq2CxYjksM/edit#gid=1394947055 which I'm planning to focus on the next days.