soar-telescope/goodman_pipeline #110: Discuss the Licence of the pipeline

simontorres, 2017-09-27:
Currently we are releasing the goodman pipeline under BSD 3-clause. We need to discuss if we still want that. GPL is the other mentioned alternative.

simontorres, 2018-02-20:
Anyone has any ideas? @b1quint @cbaorion ? I vote for GPL although we need to know if that conflicts with third-party software that we use.

cbaorion, 2018-02-20:
I don't have an informed opinion, since I am not aware of details of 
various licensing schemes. I would go for GPL because its what I am more 
familiar (if I can call it that). I would hear what Bruno has to say. He 
knows far more about this that I do.  What is the specific conflict?

Cesar

-- 
********************************************
* César Briceño                            *
* ******************************************
* Cerro Tololo Inter-American Observatory  *                                     *
* Casilla 603                              *
* La Serena 1700000, CHILE                 *
* Tlf: +56-51-2205294                      *
* Email: cbriceno@ctio.noao.edu            *
********************************************