soar-telescope/goodman_pipeline #124: Build is failing because can't connect to continuum.

simontorres, 2017-11-21:
@b1quint Could you fix the line in `.travis.yml` that is causing build to fail on travis?

```python
- if [[ "$TRAVIS_PYTHON_VERSION" == "2.7" ]]; then
    wget https://repo.continuum.io/miniconda/Miniconda2-latest-Linux-x86_64.sh -O miniconda.sh;
  else
    wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda.sh;
  fi
```

https://travis-ci.org/soar-telescope/goodman/jobs/305423785

simontorres, 2017-11-21:
Just checked the links and they both work. So it must be something else.

b1quint, 2017-11-28:
Did you manage to solve this problem?

simontorres, 2017-11-30:
Haven't touched it since.

simontorres, 2018-01-08:
This problem solved itself. Apparently was an internal failure of travis. There are other issues right now but not related to this so I'm closing this issue.

closed 2018-01-08 by simontorres