soar-telescope/goodman_pipeline #126: De-couple wavelength calibration and spectrum extraction

b1quint, 2018-02-13:
Last week I used Goodman with an astronomer using a custom angle and could not run `redspec`. He pointed that it would be **very** useful to have the extracted spectrum even without wavelength calibrated. How easy/difficult is it to de-couple these two steps?

simontorres, 2018-02-19:
It is actually very easy, it was not implemented before because I was avoiding to create too many files. It would be good to know if it is necessary to have a command line argument for switching it on or off. I left it saving the extracted files all the time because 1D files are just 50K in size.

b1quint, 2018-02-19:
That's great! I believe we could keep the extracted files for now as it is. 

My suggestion would, then, to have an argument in the command line that stops the program after extraction.

b1quint, 2018-02-19:
That's great!

So that is actually done. If you do not want to add an argument to stop the program after the extraction, is it possible to make it more obvious to the user that you saved the extracted spectrum?

simontorres, 2018-02-19:
It would be totally fine to add new arguments but eventually the amount will become very large. 
Regarding the argument for stopping the program, do you mean preventing it from attempting to obtain a wavelength solution? I can do that but for now I would like to focus on the wavelength solution and the comparison library. Maybe another issue?

I don't have any simple idea regarding  informing the user other than printing a message (already done).

b1quint, 2018-02-26:
Since @simontorres made the extracted spectrum to be always written, I would close this issue. We just need the updated version at soardata3.

simontorres, 2018-02-26:
Agreed, The updated version is already installed on soardata3 using the virtual environment *astroconda* We can later create another issue/discuss how do we want to handle control over this feature.

closed 2018-02-26 by simontorres