soar-telescope/goodman_pipeline #127: Use specutils in the pipeline.

simontorres, 2018-02-26:
@b1quint  suggested that I should use `specutils` (`specutils.wcs.specwcs`, `specutils.io.write_fits`, etc.)

I looked it up and I think that: at the moment it does not help a lot to our current problems from which one of the most important one is to have _"reliable wcs solutions"_ . 

My suggestion would be to leave this for later. It is a healthy exercise to rewrite the code once the problem has been already solved.

Waiting for Feedback.

b1quint, 2018-02-26:
`specutils` should only be used if it really helps to have the Spectral WCS easier to implement. If not, we could easily forget it.

simontorres, 2018-10-26:
This will not happen in the near future.

closed 2018-10-26 by simontorres