soar-telescope/goodman_pipeline #128: Drop Python 2.7 from the list of supported Python Versions.

simontorres, 2018-02-26:
@b1quint and I were discussion about python versions and we thought it would be a good idea to drop Python 2 out of supported python versions.
Some of the points on favor are the following. (you add your own, even against).
- Astropy will only support Python 3. [see it here](http://docs.astropy.org/en/stable/whatsnew/3.0.html#whatsnew-3-0-python3)
- The pipeline's official installation method will be using astroconda virtual environments. Users will not have to worry about having to have python 2. Also the process is quite straightforward.
- It is easier to develop for less python versions ...

cbaorion, 2018-02-26:
Agreed.

-- 
********************************************
* César Briceño                            *
* ******************************************
* Cerro Tololo Inter-American Observatory  *                                     *
* Casilla 603                              *
* La Serena 1700000, CHILE                 *
* Tlf: +56-51-2205294                      *
* Email: cbriceno@ctio.noao.edu            *
********************************************

b1quint, 2018-05-04:
Did we need to perform any action on this or this issue can be closed?

closed 2018-05-17 by simontorres