soar-telescope/goodman_pipeline #129: Parameters that should be controlable either by configuration file or by command-line arguments

simontorres, 2018-02-28:
We have already discussed that we should not just add command-line arguments because we might end up with a very large list, however we must find a way of doing this (command-line arguments included) Here is a list of parameter that may be useful to be able to modify from a user's point of view.

- Background subtraction separation from the edge of target extraction region.
- Target extraction with as a function of target's fwhm.
- Maximum number of targets to be extracted.
- DCR timeout

b1quint, 2018-05-04:
See Issue #102

simontorres, 2020-12-01:
This issue have become irrelevant. There are improvements to make but keeping this issue open might just be distraction

closed 2020-12-01 by simontorres