soar-telescope/goodman_pipeline #133: astropy.io.fits.verify.VerifyError: Unparsable card (LAMP_HGA), fix it first with .verify('fix')

b1quint, 2018-05-06:
This is the error message that has been showing during today's observation. Here is the instrumental configuration:
* Goodman Red Camera
* Grating 400 l/mm
* Wavelength Mode - Custom - Central Wavelength = 603 nm

Here is the complete traceback:
```
(astroconda) [goodman@soardata3 2018-05-06]$ redccd
[08:26:26PM][    INFO]: Starting Goodman HTS Pipeline Log
[08:26:26PM][    INFO]: Local Time    : 2018-05-06 20:26:26.986065
[08:26:27PM][    INFO]: Universal Time: 2018-05-06 23:26:27.029889
[08:26:27PM][    INFO]: Pipeline Version: 1.0.0
Traceback (most recent call last):
  File "/home/goodman/anaconda3/envs/astroconda/bin/redccd", line 12, in <module>
    GOODMAN_CCD()
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/pipeline/images/goodman_ccd.py", line 237, in __call__
    self.data_classifier(raw_path=self.args.raw_path)
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/pipeline/images/data_classifier.py", line 53, in __call__
    ifc = ImageFileCollection(self.raw_path)
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/ccdproc/image_collection.py", line 153, in __init__
    self.keywords = keywords
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/ccdproc/image_collection.py", line 302, in keywords
    self._summary = self._fits_summary(new_keys_lst)
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/ccdproc/image_collection.py", line 619, in _fits_summary
    missing_marker=missing_marker)
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/ccdproc/image_collection.py", line 530, in _dict_from_fits_header
    for k, v in six.iteritems(h):
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/astropy/io/fits/header.py", line 933, in items
    yield (card.keyword, card.value)
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/astropy/io/fits/card.py", line 284, in value
    self._value = self._parse_value()
  File "/home/goodman/anaconda3/envs/astroconda/lib/python3.5/site-packages/astropy/io/fits/card.py", line 743, in _parse_value
    ".verify('fix').".format(self.keyword))
astropy.io.fits.verify.VerifyError: Unparsable card (LAMP_HGA), fix it first with .verify('fix').
```

b1quint, 2018-05-07:
It seems that just reading and writing it again using CCDData.read and CCDData.write fixes the problem. This should be integrated into the main code, though. Example:

```python
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import os
import sys

from ccdproc import CCDData

def fix_goodman_header():

    list_of_files = sys.argv[1:]

    for filename in list_of_files:

        ccd = CCDData.read(filename, unit="adu")

        path, name = os.path.split(filename)
        filename = os.path.join(path, 'h' + name)

        ccd.write(filename, overwrite=True)

if __name__ == '__main__':
    fix_goodman_header()
```

simontorres, 2018-05-11:
I notified Stephen about this issue when it first appeared. The solution you propose is correct though I believe this should be fixed from the Goodman's side. i.e. UNC.

commit e4be89a9 (2018-08-10): Specified correct version for cython.

This is part of PR #201
Close #160
Close #133
closed 2018-08-10 by simontorres