soar-telescope/goodman_pipeline #164: Update lookup table for dcr parameters.

simontorres, 2018-07-06:
This is a required job that has to be done by hand.

The lookup table resides in `class:GenerateDcrParFile`

simontorres, 2018-07-10:
#174

Allonck, 2024-03-11:
[WIP} I will test 2x1 binning:

```terminal
(goodman_pipeline) cmontalban@lswlan59 2100_data % redccd --red-path RED_lacos --cosmic lacosmic --keep-cosmic-files 
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[17:04:48][I]: Starting Goodman HTS Pipeline Log
[17:04:48][I]: Local Time    : 2024-03-11 17:04:48.398925
[17:04:48][I]: Universal Time: 2024-03-11 20:04:48.398972
[17:04:49][I]: Pipeline Version: 1.3.7 (latest)
[17:04:49][I]: Found 50 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164/2100_data.
[17:04:49][W]: Reduction folder doesn't exist.
[17:04:49][I]: Created directory for reduced data.
[17:04:49][I]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164/2100_data/RED_lacos
[17:04:50][I]: Instrument: Red Camera
[17:04:50][I]: Detected Spectroscopy Data from Red Camera
[17:04:50][I]: Observing Technique: Spectroscopy
[17:04:50][I]: Organizing data for this configuration: Gain: 1.48, Noise: 3.89, ROI: CUST_0_4142_980_948_1x2
[17:05:01][I]: Found valid data, appending to data container list
[17:05:01][W]: Determining trim section. Assuming you have only one kind of data in this folder
[17:05:01][I]: Trim Section: [51:4110,2:948]
[17:05:01][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[17:05:01][I]: Overscan Region: [6:49,1:948]
[17:05:01][I]: Creating master bias
[17:05:01][I]: Combining 21 images to create master bias
[17:05:12][I]: Saving FITS file to master_bias_RED_SP_1x2_R03.89_G01.48.fits
[17:05:12][I]: Created master bias: master_bias_RED_SP_1x2_R03.89_G01.48.fits
[17:05:12][W]: Determining trim section. Assuming you have only one kind of data in this folder
[17:05:12][I]: Trim Section: [51:4110,2:948]
[17:05:13][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[17:05:13][I]: Overscan Region: [6:49,1:948]
[17:05:13][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[17:05:13][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[17:05:13][I]: Creating Master Flat
[17:05:13][I]: Combining 15 images to create master flat
[17:05:21][I]: Saving FITS file to master_flat_2100Custom_6500nm_GG455_0.45_dome.fits
[17:05:21][I]: Created Master Flat: RED_lacos/master_flat_2100Custom_6500nm_GG455_0.45_dome.fits
[17:05:21][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[17:05:21][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[17:05:21][I]: Creating Master Flat
[17:05:21][I]: Combining 5 images to create master flat
[17:05:24][I]: Saving FITS file to master_flat_2100Custom_6500nm_GG455_0.45_night.fits
[17:05:24][I]: Created Master Flat: RED_lacos/master_flat_2100Custom_6500nm_GG455_0.45_night.fits
[17:05:24][W]: Determining trim section. Assuming you have only one kind of data in this folder
[17:05:24][I]: Trim Section: [51:4110,2:948]
[17:05:24][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[17:05:24][I]: Overscan Region: [6:49,1:948]
[17:05:24][I]: Processing Science Target: Haro_5-2_A
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[17:05:24][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[17:05:24][I]: Creating Master Flat
[17:05:24][I]: Combining 5 images to create master flat
[17:05:26][I]: Saving FITS file to master_flat_2100Custom_6500nm_GG455_0.45_night.fits
[17:05:26][I]: Created Master Flat: RED_lacos/master_flat_2100Custom_6500nm_GG455_0.45_night.fits
[17:05:27][I]: Saving FITS file to norm_master_flat_2100Custom_6500nm_GG455_0.45_night.fits
[17:05:27][I]: Cosmic rays rejected using astroscrappy's lacosmic
[17:05:27][I]: Saving binary mask of cosmic rays to crmask_0273_Haro_5-2_B_19-11-2021_comp.fits
[17:05:27][I]: Saving FITS file to crmask_0273_Haro_5-2_B_19-11-2021_comp.fits
[17:05:27][I]: Saving image: RED_lacos/cfzst_0273_Haro_5-2_B_19-11-2021_comp.fits
[17:05:27][I]: Saving FITS file to cfzst_0273_Haro_5-2_B_19-11-2021_comp.fits
[17:05:28][I]: Cosmic rays rejected using astroscrappy's lacosmic
[17:05:28][I]: Saving binary mask of cosmic rays to crmask_0277_Haro_5-2_A_19-11-2021_comp.fits
[17:05:28][I]: Saving FITS file to crmask_0277_Haro_5-2_A_19-11-2021_comp.fits
[17:05:28][I]: Saving image: RED_lacos/cfzst_0277_Haro_5-2_A_19-11-2021_comp.fits
[17:05:28][I]: Saving FITS file to cfzst_0277_Haro_5-2_A_19-11-2021_comp.fits
[17:05:29][I]: Cosmic rays rejected using astroscrappy's lacosmic
[17:05:29][I]: Saving binary mask of cosmic rays to crmask_0278_Haro_5-2_A_19-11-2021.fits
[17:05:29][I]: Saving FITS file to crmask_0278_Haro_5-2_A_19-11-2021.fits
[... 12 lines cut ...]
[17:05:31][I]: Cosmic rays rejected using astroscrappy's lacosmic
[17:05:31][I]: Saving binary mask of cosmic rays to crmask_0281_Haro_5-2_B_19-11-2021.fits
[17:05:31][I]: Saving FITS file to crmask_0281_Haro_5-2_B_19-11-2021.fits
[17:05:31][I]: Saving image: RED_lacos/cfzst_0281_Haro_5-2_B_19-11-2021.fits
[... cut ...]
[17:05:33][I]: Saving binary mask of cosmic rays to crmask_0284_Haro_5-2_A_19-11-2021_comp.fits
[17:05:33][I]: Saving FITS file to crmask_0284_Haro_5-2_A_19-11-2021_comp.fits
[17:05:33][I]: Saving image: RED_lacos/cfzst_0284_Haro_5-2_A_19-11-2021_comp.fits
[17:05:33][I]: Saving FITS file to cfzst_0284_Haro_5-2_A_19-11-2021_comp.fits
 ```

>COMPARE OUTPUTS in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11->18_test_164/2100_data/RED_dcr/cosmicr

>AND /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11->18_test_164/2100_data/RED_lacos

Allonck, 2024-03-12:
# Binning 2x2 dcr

Using ``redccd --cosmic dcr --keep-cosmic-files --red-path RED_dcr``
```terminal
(goodman_pipeline) cmontalban@lswlan59 2021-11-18_test_164 % redccd --cosmic dcr --keep-cosmic-files --red-path RED_dcr
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[11:09:55][I]: Starting Goodman HTS Pipeline Log
[11:09:55][I]: Local Time    : 2024-03-12 11:09:55.374024
[11:09:55][I]: Universal Time: 2024-03-12 14:09:55.374073
[11:09:56][I]: Pipeline Version: 1.3.7 (latest)
[11:09:56][I]: Found 67 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164.
[11:09:56][W]: Reduction folder doesn't exist.
[11:09:56][I]: Created directory for reduced data.
[11:09:56][I]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164/RED_dcr
[11:09:57][I]: Instrument: Red Camera
[11:09:57][I]: Detected Spectroscopy Data from Red Camera
[11:09:57][I]: Observing Technique: Spectroscopy
[11:09:57][I]: Organizing data for this configuration: Gain: 1.48, Noise: 3.89, ROI: Spectroscopic 2x2
[11:10:08][I]: Found valid data, appending to data container list
[11:10:08][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:10:08][I]: Trim Section: [26:2055,2:948]
[11:10:08][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:10:08][I]: Overscan Region: [3:24,1:948]
[11:10:08][I]: Creating master bias
[11:10:08][I]: Combining 21 images to create master bias
[11:10:13][I]: Saving FITS file to master_bias_RED_SP_2x2_R03.89_G01.48.fits
[11:10:13][I]: Created master bias: master_bias_RED_SP_2x2_R03.89_G01.48.fits
[11:10:13][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:10:13][I]: Trim Section: [26:2055,2:948]
[11:10:13][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:10:13][I]: Overscan Region: [3:24,1:948]
[11:10:13][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[11:10:13][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[11:10:13][I]: Creating Master Flat
[11:10:14][I]: Combining 15 images to create master flat
[11:10:17][I]: Saving FITS file to master_flat_400m2_GG455_1.0_dome.fits
[11:10:17][I]: Created Master Flat: RED_dcr/master_flat_400m2_GG455_1.0_dome.fits
[11:10:17][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[11:10:18][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[11:10:18][I]: Creating Master Flat
[11:10:18][I]: Combining 15 images to create master flat
[11:10:22][I]: Saving FITS file to master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
[11:10:22][I]: Created Master Flat: RED_dcr/master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
[11:10:22][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:10:22][I]: Trim Section: [26:2055,2:948]
[11:10:22][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:10:22][I]: Overscan Region: [3:24,1:948]
[11:10:22][I]: Processing Science Target: Haro_5-2_A
[11:10:22][I]: Attempting to find a suitable Master Flat
[11:10:22][E]: There is no flat available
[11:10:22][C]: Failed to obtain master flat
[11:10:22][I]: Master flat inexistent, can't find slit trim section
[11:10:22][W]: The file 0259_Haro_5-2_A_19-11-2021_comp.fits will not be flatfielded
[11:10:22][W]: DCR does apply the correction to images if you want the mask use --keep-cosmic-files
[11:10:22][I]: Saving FITS file to zt_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:10:22][I]: Saving image: RED_dcr/zt_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:10:22][I]: Removing cosmic rays using DCR by Wojtek Pych
[11:10:22][E]: b'RED_dcr/zt_0259_Haro_5-2_A_19-11-2021_comp.fits: No such file or directory\n'
[11:10:22][W]: Cosmic ray rejection returned a None.
[11:10:22][W]: The file 0260_Haro_5-2_A_19-11-2021.fits will not be flatfielded
[11:10:22][W]: DCR does apply the correction to images if you want the mask use --keep-cosmic-files
[11:10:22][I]: Saving FITS file to zt_0260_Haro_5-2_A_19-11-2021.fits
[11:10:22][I]: Saving image: RED_dcr/zt_0260_Haro_5-2_A_19-11-2021.fits
[11:10:22][I]: Removing cosmic rays using DCR by Wojtek Pych
[11:10:22][E]: b'RED_dcr/zt_0260_Haro_5-2_A_19-11-2021.fits: No such file or directory\n'
[11:10:22][W]: Cosmic ray rejection returned a None.
[11:10:22][W]: The file 0261_Haro_5-2_A_19-11-2021.fits will not be flatfielded
[... 115 lines cut ...]

``` terminal
dcr.par
master_bias_RED_SP_2x2_R03.89_G01.48.fits
master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
master_flat_400m2_GG455_1.0_dome.fits
zt_0255_Haro_5-2_A_19-11-2021_comp.fits
zt_0256_Haro_5-2_A_19-11-2021.fits
zt_0257_Haro_5-2_A_19-11-2021.fits
zt_0258_Haro_5-2_A_19-11-2021.fits
zt_0259_Haro_5-2_A_19-11-2021_comp.fits
zt_0260_Haro_5-2_A_19-11-2021.fits
zt_0261_Haro_5-2_A_19-11-2021.fits
zt_0262_Haro_5-2_A_19-11-2021.fits
zt_0263_Haro_5-2_B_19-11-2021.fits
zt_0264_Haro_5-2_B_19-11-2021.fits
zt_0265_Haro_5-2_B_19-11-2021.fits
zt_0266_Haro_5-2_B_19-11-2021_comp.fits
zt_0272_Haro_5-2_B_19-11-2021_comp.fits
zt_0274_Haro_5-2_B_19-11-2021.fits
zt_0275_Haro_5-2_B_19-11-2021.fits
zt_0276_Haro_5-2_B_19-11-2021.fits
(goodman_pipeline) cmontalban@lswlan59 RED_dcr % 
```
I obtain the same result without using the keep files option: ``redccd --cosmic dcr --red-path RED_dcr_without_keeping_mask_files``

Allonck, 2024-03-12:
# Binning 2x2 Lacosmic:

Using ``redccd --cosmic lacosmic --keep-cosmic-files --red-path RED_lacosmic``

``` terminal
(goodman_pipeline) cmontalban@lswlan59 2021-11-18_test_164 % redccd --cosmic lacosmic --keep-cosmic-files --red-path RED_lacosmic
/Users/cmontalban/anaconda3/envs/goodman_pipeline/bin/redccd:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  __import__('pkg_resources').require('goodman-pipeline==1.3.7')
[11:19:44][I]: Starting Goodman HTS Pipeline Log
[11:19:44][I]: Local Time    : 2024-03-12 11:19:44.071152
[11:19:44][I]: Universal Time: 2024-03-12 14:19:44.071202
[11:19:46][I]: Pipeline Version: 1.3.7 (latest)
[11:19:46][I]: Found 67 fits files in /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164.
[11:19:46][W]: Reduction folder doesn't exist.
[11:19:46][I]: Created directory for reduced data.
[11:19:46][I]: New directory: /Users/cmontalban/Desktop/SOARdata/Haro_5-2/2021-11-18/2021-11-18_test_164/RED_lacosmic
[11:19:46][I]: Instrument: Red Camera
[11:19:46][I]: Detected Spectroscopy Data from Red Camera
[11:19:46][I]: Observing Technique: Spectroscopy
[11:19:47][I]: Organizing data for this configuration: Gain: 1.48, Noise: 3.89, ROI: Spectroscopic 2x2
[11:19:57][I]: Found valid data, appending to data container list
[11:19:57][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:19:57][I]: Trim Section: [26:2055,2:948]
[11:19:57][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:19:57][I]: Overscan Region: [3:24,1:948]
[11:19:57][I]: Creating master bias
[11:19:58][I]: Combining 21 images to create master bias
[11:20:03][I]: Saving FITS file to master_bias_RED_SP_2x2_R03.89_G01.48.fits
[11:20:03][I]: Created master bias: master_bias_RED_SP_2x2_R03.89_G01.48.fits
[11:20:03][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:20:03][I]: Trim Section: [26:2055,2:948]
[11:20:03][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:20:03][I]: Overscan Region: [3:24,1:948]
[11:20:03][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[11:20:03][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[11:20:03][I]: Creating Master Flat
[11:20:03][I]: Combining 15 images to create master flat
[11:20:07][I]: Saving FITS file to master_flat_400m2_GG455_1.0_dome.fits
[11:20:07][I]: Created Master Flat: RED_lacosmic/master_flat_400m2_GG455_1.0_dome.fits
[11:20:07][W]: Only flats found in this group
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[11:20:07][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[11:20:07][I]: Creating Master Flat
[11:20:07][I]: Combining 15 images to create master flat
[11:20:11][I]: Saving FITS file to master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
[11:20:11][I]: Created Master Flat: RED_lacosmic/master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
[11:20:11][W]: Determining trim section. Assuming you have only one kind of data in this folder
[11:20:11][I]: Trim Section: [26:2055,2:948]
[11:20:11][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[11:20:11][I]: Overscan Region: [3:24,1:948]
[11:20:11][I]: Processing Science Target: Haro_5-2_A
[11:20:11][I]: Attempting to find a suitable Master Flat
[11:20:11][E]: There is no flat available
[11:20:11][C]: Failed to obtain master flat
[11:20:11][I]: Master flat inexistent, can't find slit trim section
[11:20:11][W]: The file 0259_Haro_5-2_A_19-11-2021_comp.fits will not be flatfielded
[11:20:11][I]: Cosmic rays rejected using astroscrappy's lacosmic
[11:20:11][I]: Saving binary mask of cosmic rays to crmask_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:20:11][I]: Saving FITS file to crmask_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:20:11][I]: Saving image: RED_lacosmic/czt_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:20:11][I]: Saving FITS file to czt_0259_Haro_5-2_A_19-11-2021_comp.fits
[11:20:11][W]: The file 0260_Haro_5-2_A_19-11-2021.fits will not be flatfielded
[11:20:12][I]: Cosmic rays rejected using astroscrappy's lacosmic
[11:20:12][I]: Saving binary mask of cosmic rays to crmask_0260_Haro_5-2_A_19-11-2021.fits
[11:20:12][I]: Saving FITS file to crmask_0260_Haro_5-2_A_19-11-2021.fits
[11:20:12][I]: Saving image: RED_lacosmic/czt_0260_Haro_5-2_A_19-11-2021.fits
[11:20:12][I]: Saving FITS file to czt_0260_Haro_5-2_A_19-11-2021.fits
[11:20:12][W]: The file 0261_Haro_5-2_A_19-11-2021.fits will not be flatfielded
[11:20:12][I]: Cosmic rays rejected using astroscrappy's lacosmic
[... 118 lines cut ...]
czt_0255_Haro_5-2_A_19-11-2021_comp.fits
czt_0256_Haro_5-2_A_19-11-2021.fits
czt_0257_Haro_5-2_A_19-11-2021.fits
czt_0258_Haro_5-2_A_19-11-2021.fits
czt_0259_Haro_5-2_A_19-11-2021_comp.fits
czt_0260_Haro_5-2_A_19-11-2021.fits
czt_0261_Haro_5-2_A_19-11-2021.fits
czt_0262_Haro_5-2_A_19-11-2021.fits
czt_0263_Haro_5-2_B_19-11-2021.fits
czt_0264_Haro_5-2_B_19-11-2021.fits
czt_0265_Haro_5-2_B_19-11-2021.fits
czt_0266_Haro_5-2_B_19-11-2021_comp.fits
czt_0272_Haro_5-2_B_19-11-2021_comp.fits
czt_0274_Haro_5-2_B_19-11-2021.fits
czt_0275_Haro_5-2_B_19-11-2021.fits
czt_0276_Haro_5-2_B_19-11-2021.fits
master_bias_RED_SP_2x2_R03.89_G01.48.fits
master_flat_400Custom_5053nm_NO_FILTER_1.0_dome.fits
master_flat_400m2_GG455_1.0_dome.fits
(goodman_pipeline) cmontalban@lswlan59 RED_lacosmic % 
```

It works pretty good. I also tried ``redccd --cosmic lacosmic  --red-path RED_lacosmic_without_keeping_mask_files`` and it works well too. 

__Therefore, the ``dcr`` option doesn't process the images__

Allonck, 2024-03-12:
# Binning 2x2 Comparison

Since the terminal usage of the pipeline didn't work as expected I used Jupyter notebook to compare the defaults ``dcr.par`` against ``Lacosmic``.

If I divide both spectrums (LACOSMIC / DCR) I get very similar results:

- For Haro 5-2 A: 
[image]

- For Haro 5-2 B:
[image]

So, my preliminar thoughts are that ``dcr`` can be used with the default parameters with 2x2 binnings images too.

Allonck, 2024-03-12:
The test with the threshold parameter seems to indicate that it is one of the critical parameters to adjust, along with the size of the BOX (X and Y). At least, upon visual inspection, the vast majority of cosmic rays disappeared when this parameter was set higher.

The value of 5 is awful, too much higher.

[image]

The default of 4 is very nice, but a large cosmic ray around X= 1500 not was eliminated.

[image]

If I down the parameter to 3, it seems to delete more consistently the ray:

[image]

__After reducing the spectra and comparing again, I'm going to have a better conclusion. Because maybe setting the parameter too low can remove part of the object spectra.__

Allonck, 2024-03-12:
# Comparison with ``thresh`` `` dcr`` parameter varying. 

Following the previous process of dividing spectra:

- For Haro 5-2 A:

[image]

- For Haro 5-2 B:

[image]
 
Therefore, by varying the threshold parameter, the best result is obtained with the default. As I thought, decreasing the threshold value decreases the cosmic rays, but also affects the spectrum. In such a case, it would be good to check the accuracy of this by varying the dimensional parameter of the box (X and Y values). If the parameters are also adequate I would have more reason to think that the dcr method can be almost as efficient as LACOSMIC.

Allonck, 2024-03-13:
Related to #388