soar-telescope/goodman_pipeline #168: LACosmic usability issues

simontorres, 2018-07-09:
Though LACosmic produced inferior results with respect to DCR we decided to keep it but there have been many changes to the pipeline since then. Now I find that LACosmic is not usable in the current implementation here are the reason and the possible fixes.

## Problem
LACosmic does not apply a correction to the images, instead it creates a mask attribute and the file is stored as a multi-extension-fits file. In older version this mask was used to mask-out those pixels during spectra extraction.

### Solution 1
Implement (or find out if already exists) a method to apply the correction to the masked pixels by some interpolation method so that we get a  corrected single extension file

### Solution 2
Implement ability to handle multi extension files and with it also implement the use of masks for using during extraction.

In general, DCR and LACosmic are two different worlds that would be ideal to have them both working at their full potential but the amount of work required to do that is large and it will also add a lot of complexity. So I'm more inclined to prefer solution 1. 

Please comment your thoughts.

b1quint, 2018-07-09:
From the `ccdproc.cosmicray_lacosmic`:

```
returns:

    nccd (~astropy.nddata.CCDData or numpy.ndarray) – An object of the same type as ccd is returned. If it is a ~astropy.nddata.CCDData, the mask attribute will also be updated with areas identified with cosmic rays masked.

    crmask (numpy.ndarray) – If an numpy.ndarray is provided as ccd, a boolean ndarray with the cosmic rays identified will also be returned.
```

That means that, if you use LaCosmic with a numpy.ndarray as input, you will have two numpy.ndarray as output. Then, you can ignore the mask. On the code snipped below extracted from the SAMI pipeline, you can see how to ignore the second returned argument:

https://github.com/soar-telescope/sami/blob/65991a6a2e2162baaa6d26806c585f1b68f2e1eb/soar_sami/data_reduction/merge.py#L765-L795

simontorres, 2018-07-09:
You can also set `ccd.mask = None` and get the same result. but the corrections are not applied to the image, unless I'm missing something.

b1quint, 2018-07-09:
So, I would try to extract the data in `numpy.array`, run lacosmic, and replace the corrected data. Something like:

```python
       data = ccd.data
       cr_data, _ = ccdproc.cosmicray_lacosmic(data, ...)
       ccd.data = cr_data
```

simontorres, 2018-07-12:
We will forget about `ccdproc`'s `lacosmic` now (most likely for ever) and focus on `astroscrappy` #171

closed 2018-07-12 by simontorres