soar-telescope/goodman_pipeline #183: RTD user manual content rearrangement

simontorres, 2018-07-12:
In my opinion (@b1quint  agrees) the information published in ReadTheDocs need some rearrangement/editing 

Here is @b1quint 's suggestion:

```
|--- About
|   |--- Features
|   |--- License
|   |--- Release Notes
|   |--- Author / Contributors
|--- Install
|   |--- Requirements
|   |--- Using Conda
|   |--- Using PIP
|   |--- Classic Method
|--- Usage
|--- Examples

```
I would like to propose the following general guidelines.

- Don't write anything before every piece already existing has been relocated.
- The focus should be what a new user might find most useful.

cbaorion, 2018-07-13:
We discussed the structure of the manual during our Friday Jul 13 meeting.
I argue for this structure for the manual. I argue it should be oriented for the users who will connect to the soar data reduction machine and reduce his spectra there, and only as a secondary goal toward people that will download and install the software.
Simon will be working more in detail on rearranging the manual and its structure, and we will revisit this on our next meeting of Monday Jul 23.

```
|--- Overview
|---- Usage
|   |---- Data requirements
|   |---- Processing your 2-D images
|   |---- Extracting the spectra
|   |---- (Cosmic ray removal)
|   |--- Examples
|--- Install
|   |--- Requirements
|   |--- Using Conda
|   |--- Using PIP
|   |--- Classic Method
|--- About 
|    |--- Licensing
|    |--- Release Notes
|    |--- Author / Contributors

```

simontorres, 2018-07-23:
This issue is solved by PR #193 Additional changes need to be handled separately

closed 2018-07-23 by simontorres