soar-telescope/goodman_pipeline #189: Accept more than one reference lamps

wschoenell, 2018-07-15:
When doing automatic wavelength calibration, I know I can choose a different folder for reference lamps. For each configuration, `redspec` expects a single reference file. When I add another reference file for a given configuration, the program gives me an error message saying that does not know which one to use.

I believe `redspec` should be able to decide which reference file to use based on the closest date of the data being reduced.

simontorres, 2018-07-17:
I think it is not necessary, at least I don't see how this can help improve the method. As the name says the reference wavelength solution is just a reference. The wavelength solution will only depend on the comparison lamp obtained for the science target. The only case when the reference lamp solution may affect is in the case where the solution itself is wrong and this is why the reference lamps that we are providing are double or even "triple-checked". Just for clarity, the reference lamp is a comparison lamp which has been calibrated in wavelength, i.e., each line has been carefully identified in a manual, interactive way. This reference lamp is fixed given the specific elements, e.g., HgAr or HgArNe. It does not change with time nor with position of the telescope, and does not need to be replaced.

simontorres, 2020-03-13:
The current solution is to build a  wavelength calibrated spectrum with each reference lamp. The ideal solution would be to combine them but there are several strategies for doing it so,

- we can't implement them all
- choosing one to implement has not been study.