soar-telescope/goodman_pipeline #207: core should not import from other pipeline's subpackages

simontorres, 2018-08-13:
Currently the module `goodman/pipeline/core/core.py` is intended to host all _independent_ functions, basic or advanced, i.e. functions that need a generic input to generate a generic output. Other modules and pipeline's subpackages should import from it but not the other way around.

For instance I'm trying to check that the version is the latest and I have to import it from `pipeline/tools/check_version.py` but there is also `pipeline/tools/reference_lamp_factory/create_reference_lamps.py` which imports functions from `core`. When I import check version this is done through `pipeline/tools/__init__.py` which at the same time imports `create_reference_lamps.py` and it fails.

I will work on a Pull Request.

closed 2018-08-13 by simontorres

cbaorion, 2018-08-13:
Hi Simon,
I am not sure why this was assigned to me.

Cesar

simontorres, 2018-08-13:
Hi Cesar,

I put you both just in case you had an important comment about it.

commit 783dad16 (2018-08-13): Merge pull request #208 from simontorres/version_check_and_athorized_API_access

Implemented Authorized GITHUB API access and  version check at start up.
closes #207