soar-telescope/goodman_pipeline #213: Add clean exit when data from both cameras are present.

simontorres, 2018-08-21:
```

(goodman) ma021432:02022018_DATA user$ redccd
[15:23:43][I]: Starting Goodman HTS Pipeline Log
[15:23:43][I]: Local Time    : 2018-08-21 15:23:43.620992
[15:23:43][I]: Universal Time: 2018-08-21 18:23:43.621069
[15:23:43][I]: Pipeline Version: 1.1.0
[15:23:47][W]: Camera changes are forbidden during the night
Traceback (most recent call last):
  File "/Users/user/anaconda3/envs/goodman/bin/redccd", line 4, in <module>
    __import__('pkg_resources').run_script('pipeline==1.1.0', 'redccd')
  File "/Users/user/anaconda3/envs/goodman/lib/python3.6/site-packages/pkg_resources/__init__.py", line 657, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/Users/user/anaconda3/envs/goodman/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1437, in run_script
    exec(code, namespace, namespace)
  File "/Users/user/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/EGG-INFO/scripts/redccd", line 12, in <module>
    GOODMAN_CCD()
  File "/Users/user/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/pipeline/images/goodman_ccd.py", line 207, in __call__
    self.data_classifier(raw_path=self.args.raw_path)
  File "/Users/user/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/pipeline/images/data_classifier.py", line 64, in __call__
    self.log.info('Instrument: ' + self.instrument + ' Camera')
TypeError: must be str, not NoneType

```

simontorres, 2018-08-21:
OK. I will have a look and let you know.

simontorres, 2018-08-21:
Just to clarify even further you need to change the value for ``OBJECT``:

``HgAr Ne_14sdss_m2`` to ``HgArNe``
``HgAr Ne_14sdss`` to ``HgArNe``

and so on.

cbaorion, 2018-08-22:
Maybe we need to make it more clear in the manual? I will check what we 
say about lamps in the manual.

-- 
********************************************
* César Briceño                            *
* ******************************************
* Cerro Tololo Inter-American Observatory  *                                     *
* Casilla 603                              *
* La Serena 1700000, CHILE                 *
* Tlf: +56-51-2205294                      *
* Email: cbriceno@ctio.noao.edu            *
********************************************

simontorres, 2018-08-22:
The best solution to this is to eliminate the user part here. We should finish the lamp keywords #135 and will never have to worry about this. We should leave the current method for backwards compatibility.

simontorres, 2018-08-22:
It should not be necessary to separate the files. I can take a closer look tomorrow, but if the pipeline is not matching the master bias and flats is because they indeed do not match. I could not find anything obvious with a quick look. Could you please e-mail me a txt list of files being used in your reduction run?

jemejia, 2018-08-23:
Hi Simón,

Attached you find the list of files I am using for the m2 camera and the m1
camera.  In both cases it creates a master_flat but it then says, or some
objects, that it is not available (please fin also attached the full output
with m2):

(goodman) ma021432:02022018_DATA jmejiar$ redccd
[11:49:15][I]: Starting Goodman HTS Pipeline Log
[11:49:15][I]: Local Time    : 2018-08-23 11:49:15.690332
[11:49:15][I]: Universal Time: 2018-08-23 14:49:15.690441
[11:49:15][I]: Pipeline Version: 1.1.0
[11:49:18][I]: Instrument: Red Camera
[11:49:18][I]: Detected Spectroscopy Data from Red Camera
[11:49:18][I]: Observing Technique: Spectroscopy
[11:49:18][I]: No special reduced data path defined. Proceeding with
defaults.
[11:49:18][W]: Reduction folder doesn't exist.
[11:49:18][I]: Created reduced data directory!
[11:49:18][I]:
/Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED
[11:49:20][W]: Ignoring all Imaging data. Assuming they are all acquisition
exposures
[11:49:21][W]: Determining trim section. Assuming you have only one kind of
data in this folder
[11:49:21][I]: Trim Section: [26:2055,2:948]
[11:49:21][I]: Overscan regions has been tested for ROI Spectroscopic 1x1,
2x2 and 3x3
[11:49:21][I]: Overscan Region: [3:24,1:948]
[11:49:21][I]: Creating master bias
[11:49:31][I]: Created master bias:
/Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/master_bias.fits
[11:49:31][I]: Creating Master Flat
[11:49:41][I]: Created Master Flat:
/Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/master_flat_400m2_GG455_1.0_dome.fits
[11:49:41][I]: Processing Science Target: 14sdss_ADQ
[11:49:41][I]: Attempting to find a suitable Master Flat
[11:49:41][E]: There is no flat available
[11:49:41][C]: Failed to obtain master flat
[11:49:41][I]: Master flat inexistent, cant find slit trim section
[11:49:41][W]: The file 0168_14SDSS_ADQ.fits will not be flatfielded

Thanks if advande for your help!

Julián

-- 

*Julian Mejia-Restrepo - Postdoctoral Fellow*

   +     | European Southern Observatory
+ E S +  | Alonso de Cordova 3107
<https://maps.google.com/?q=Alonso+de+Cordova+3107&entry=gmail&source=g>
   O     | Vitacura, Santiago
   +     | Chile
jmejiar@eso.org

0111_dflat_400m2_5s_70p.fits
0112_dflat_400m2_5s_70p.fits
0113_dflat_400m2_5s_70p.fits
0114_dflat_400m2_5s_70p.fits
0115_dflat_400m2_5s_70p.fits
0116_dflat_400m2_5s_70p.fits
0117_dflat_400m2_5s_70p.fits
0118_dflat_400m2_5s_70p.fits
0119_dflat_400m2_5s_70p.fits
0120_dflat_400m2_5s_70p.fits
0121_dflat_400m2_5s_70p.fits
0122_dflat_400m2_5s_70p.fits
0123_dflat_400m2_5s_70p.fits
[... 472 lines cut ...]
[11:50:53][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Cosmic rays rejected using astroscrappy's lacosmic
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0190_00SDSS_spec_m2.fits
[11:50:56][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Changing cosmic ray rejection method from 'lacosmic' to 'none' for comparison lamp. Prefix 'c' will be added anyway.
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0191_00SDSS_spec_m2.fits
[11:50:56][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Changing cosmic ray rejection method from 'lacosmic' to 'none' for comparison lamp. Prefix 'c' will be added anyway.
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0192_00SDSS_spec_m2.fits
[11:50:56][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Changing cosmic ray rejection method from 'lacosmic' to 'none' for comparison lamp. Prefix 'c' will be added anyway.
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0193_00SDSS_spec_m2.fits
[11:50:56][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Changing cosmic ray rejection method from 'lacosmic' to 'none' for comparison lamp. Prefix 'c' will be added anyway.
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0194_00SDSS_spec_m2.fits
[11:50:56][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:56][I]: Changing cosmic ray rejection method from 'lacosmic' to 'none' for comparison lamp. Prefix 'c' will be added anyway.
[11:50:56][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0195_00SDSS_spec_m2.fits
[11:50:57][I]: Processing Science Target: LTT2415
[11:50:57][I]: Attempting to find a suitable Master Flat
INFO: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file. [astropy.nddata.ccddata]
[11:50:57][I]: using the unit adu passed to the FITS reader instead of the unit adu in the FITS file.
[11:50:57][I]: Setting cosmic ray rejection method to: lacosmic
[11:50:59][I]: Cosmic rays rejected using astroscrappy's lacosmic
[11:50:59][I]: Saving image: /Users/jmejiar/Dropbox/Astrophysics/AGNs/observations/INFRARED_LARGE_PROGRAM/02022018_DATA/RED/cfzsto_0215_LTT2415_spec_m2.fits

simontorres, 2018-08-23:
I just run ``redccd`` to the data with the files of your list.

The following files are not flat fielded and as I said before this is OK, because they are not supposed to be flatfielded.
```
0168_14SDSS_ADQ.fits
0169_14SDSS_ADQ.fits
0170_14SDSS_ADQ.fits
0171_14SDSS_ADQ.fits
0172_14SDSS_ADQ.fits
0173_14SDSS_ADQ.fits
0174_14SDSS_ADQ.fits
0209_LTT2415_LAMP_m2.fits
0210_LTT2415_LAMP_m2.fits
0211_LTT2415_LAMP_m2.fits
0212_LTT2415_LAMP_m2.fits
0213_LTT2415_LAMP_m2.fits
0214_LTT2415_spec_m2.fits
```
Please observe the following screenshot.

[image]

The files from 168 to 174 are slit acquisition images because the grating is not IN and the camera and grating angle are zero.

The second group of images (209 to 214) the observer forgot to put the order blocking filter ``GG-455``.

Also note that Goodman HTS has two cameras: ``Red`` and ``Blue``. ``m1`` and ``m2`` in this case are _spectroscopic modes_ of the 400 l/mm grating.
http://www.ctio.noao.edu/soar/sites/default/files/GOODMAN/Images/New_Goodman_Cheat_Sheet.pdf
https://goodman.readthedocs.io/en/latest/

simontorres, 2018-08-23:
Fixed on #214

closed 2018-08-23 by simontorres

jemejia, 2018-08-24:
Thanks Simon,

I know m2 and m1 are spectroscopic modes but there might be an error in the
pipeline because if you try to run redccd without splitting them, redccd
complains about using two different cameras.

Cheers,

Julián

simontorres, 2018-08-24:
If you refer to this traceback

```
(goodman) ma021432:02022018_DATA jmejiar$ redccd
[15:23:43][I]: Starting Goodman HTS Pipeline Log
[15:23:43][I]: Local Time    : 2018-08-21 15:23:43.620992
[15:23:43][I]: Universal Time: 2018-08-21 18:23:43.621069
[15:23:43][I]: Pipeline Version: 1.1.0
[15:23:47][W]: Camera changes are forbidden during the night
Traceback (most recent call last):
  File "/Users/jmejiar/anaconda3/envs/goodman/bin/redccd", line 4, in <module>
    __import__('pkg_resources').run_script('pipeline==1.1.0', 'redccd')
  File "/Users/jmejiar/anaconda3/envs/goodman/lib/python3.6/site-packages/pkg_resources/__init__.py", line 657, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/Users/jmejiar/anaconda3/envs/goodman/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1437, in run_script
    exec(code, namespace, namespace)
  File "/Users/jmejiar/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/EGG-INFO/scripts/redccd", line 12, in <module>
    GOODMAN_CCD()
  File "/Users/jmejiar/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/pipeline/images/goodman_ccd.py", line 207, in __call__
    self.data_classifier(raw_path=self.args.raw_path)
  File "/Users/jmejiar/anaconda3/envs/goodman/lib/python3.6/site-packages/pipeline-1.1.0-py3.6.egg/pipeline/images/data_classifier.py", line 64, in __call__
    self.log.info('Instrument: ' + self.instrument + ' Camera')
TypeError: must be str, not NoneType
```
There is indeed a bug which is basically that a ``None`` object can't be converted to string automatically. This has been solved. The reason why ``self.instrument`` is ``None`` is because when the pipeline groups all values from the keyword ``INSTCONF`` it founds more than one (from maximum two) and it should be just one. I will check again your data and will let you know.

Also there is a new release (version 1.1.1) which include several improvements. https://goodman.readthedocs.io/en/latest/change_history.html#v1-1-1-23-08-2018

simontorres, 2018-08-24:
I double checked the data and I did not get the error of the _double camera_, so I'm thinking you might have mixed data.