soar-telescope/goodman_pipeline #254: Installation in TCSH - instructions

cbaorion, 2018-10-29:
This is a guide for installing the goodman_pipeline when using tcsh/csh instead of bash, The information is adapted from the excellent explanation in ESMValGroup/ESMValTool#301

-   Install Miniconda3 in your desired location (say` <prefix>/miniconda3/)`

- Do not prepend the Miniconda3 install location to PATH in your .cshrc file (as you would do with the standard installation for bash), but instead  add the following line to your .cshrc file: `source <prefix>/miniconda3/etc/profile.d/conda.csh`
- `source ~/.cshrc`
- create the goodman_pipeline environment:  `conda env create --file environment.yml python=3.6`. In this example, we are forcing the system to use python 3.6., but it should also work if you do not specify the python version but simply write `conda env create --file environment.yml `
- activate the environment using `conda activate goodman_pipeline` instead of `source activate goodman_pipeline`
- cd to the folder where you installed the goodman_pipeline: `cd goodman_pipeline-<latest tag>`

    continue with the installation of DCR as outlined in the installation guide

simontorres, 2019-08-29:
This should go into the readthedocs page

simontorres, 2024-03-06:
This issue does not need to be open

closed 2024-03-06 by simontorres