soar-telescope/goodman_pipeline #320: Imaging data wrongly identified as spectroscopic observations

dsanmartim, 2020-04-12:
I've obtained some imaging data and I was trying to use the redccd pipeline to get the ccd reduction done, but the pipeline is wrongly identifying the data as

**[15:00:19][I]: Detected Spectroscopy Data from Red Camera
[15:00:19][I]: Observing Technique: Spectroscopy**

which is wrong. All data inside the raw_path was observed in imaging mode. Below you can find some relevant information to try to understand the issue.

REDCCD pipeline was run using the following parameters:

$> raw_path="/path_to_night/20191220/GOODMAN/RAW/"
$> red_path="/path_to_night/20191220/GOODMAN/REDCCD/"
$> redccd --raw-path "$raw_path" --red-path "$red_path" --flat-normalize mean

I'm using the Goodman HTS Pipeline v1.3.0.

This is the  redccd log output:

_Failed to import any qt binding
[15:00:16][I]: Starting Goodman HTS Pipeline Log
[15:00:16][I]: Local Time    : 2020-04-12 15:00:16.745594
[15:00:16][I]: Universal Time: 2020-04-12 19:00:16.745744
[15:00:17][I]: Pipeline Version: 1.3.0 (latest)
[15:00:17][I]: Found 158 fits files in /path_to_data/20191220/GOODMAN/RAW.
[15:00:17][I]: Folder for reduced data is empty (good).
[15:00:19][I]: Instrument: Red Camera
**[15:00:19][I]: Detected Spectroscopy Data from Red Camera
[15:00:19][I]: Observing Technique: Spectroscopy**
[15:00:21][I]: Organizing data for this configuration: Gain: 3.78, Noise: 8.99, ROI: Imaging 1x1
[15:00:22][I]: Found valid data, appending to data container list
[15:00:22][W]: Determining trim section. Assuming you have only one kind of data in this folder
[15:00:22][I]: Trim Section: [51:4110,2:3096]
[15:00:22][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[15:00:22][I]: Overscan Region: [6:49,1:3096]
[15:00:22][I]: Creating master bias
[15:01:15][I]: Saving FITS file to master_bias_RED_SP_1x1_R08.99_G03.78.fits
[15:01:15][I]: Created master bias: master_bias_RED_SP_1x1_R08.99_G03.78.fits
[15:01:15][W]: Determining trim section. Assuming you have only one kind of data in this folder
[15:01:15][I]: Trim Section: [51:4110,2:3096]
[15:01:15][I]: Overscan regions has been tested for ROI Spectroscopic 1x1, 2x2 and 3x3
[15:01:15][I]: Overscan Region: [6:49,1:3096]
Traceback (most recent call last):
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/bin/redccd", line 4, in <module>
    __import__('pkg_resources').run_script('goodman-pipeline==1.3.0', 'redccd')
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1464, in run_script
    exec(code, namespace, namespace)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/EGG-INFO/scripts/redccd", line 17, in <module>
    GOODMAN_CCD()
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/images/goodman_ccd.py", line 262, in __call__
    process_images()
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/images/image_processor.py", line 134, in __call__
    morning_twilight=self.morning_twilight)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/core/core.py", line 2233, in name_master_flats
    wavmode = spectroscopic_mode(header=header)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/core/core.py", line 4072, in __call__
    blocking_filter=blocking_filter)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/core/core.py", line 4118, in get_mode
    cam_ang=camera_targ)
  File "/Users/dsanmartim/opt/anaconda3/envs/goodman_pipeline/lib/python3.6/site-packages/goodman_pipeline-1.3.0-py3.6.egg/goodman_pipeline/core/core.py", line 1443, in get_central_wavelength
    grating_frequency = float(grating) / u.mm
ValueError: could not convert string to float:_

And here you can see a few relevant keywords from the raw data:

--> hsel *fits $I,INSTCONF,GRATING,SLIT,WAVMODE,OBSTYPE,ROI yes
0084_flat_R.fits	Red	NO_GRATING	NO_MASK	IMAGING	FLAT	"Imaging 1x1"
0085_flat_R.fits	Red	NO_GRATING	NO_MASK	IMAGING	FLAT	"Imaging 1x1"
0086_flat_R.fits	Red	NO_GRATING	NO_MASK	IMAGING	FLAT	"Imaging 1x1"
0087_flat_R.fits	Red	NO_GRATING	NO_MASK	IMAGING	FLAT	"Imaging 1x1"
[... 134 lines cut ...]
0236_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0237_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0240_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0241_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0242_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0243_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0244_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0245_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0246_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0247_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0248_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0249_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0250_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0251_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0254_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0255_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0256_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0257_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0258_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"
0259_SO2019B-019.fits	Red	NO_GRATING	NO_MASK	IMAGING	OBJECT	"Imaging 1x1"

Let me know if you need additional information.

Thanks.
David.

simontorres, 2020-04-13:
I will take a look at it today.

simontorres, 2020-04-13:
@dsanmartim thanks for posting this issue, it is a bug indeed and though I have located the source of the problem I will no be able to provide a solution today. I'm working on getting a more robust solution so it might take one or two days. Feel free to remind me in case you don't get a response soon.

dsanmartim, 2020-04-14:
Hi @simontorres. Thanks for the update. I'm glad this will be useful to debug the code.

simontorres, 2020-04-15:
@dsanmartim I think I got it working properly, please re-install the code as follows.

`pip install goodman-pipeline==1.3.1.dev5`

Let me know how it goes.

dsanmartim, 2020-04-16:
Hi Simon. It's working fine now. Thanks a lot!

simontorres, 2020-04-16:
Excellent,
 I will close this issue now, if there is anything related to this same issue feel free to reopen it or comment.

closed 2020-04-16 by simontorres