soar-telescope/goodman_pipeline #335: Check which routine is used to identify technique

simontorres, 2020-11-02:
There are at least two

Allonck, 2024-03-11:
At least in the ``core.py`` script, there is only one routine to identify technique. It begins on line 1956.

``` python

def identify_technique(target, obstype, slit, grating, wavmode, roi):
    """Identify whether is Imaging or Spectroscopic data

    Args:
        target (str): Target name as in the keyword `OBJECT` this is useful in
        Automated aquisition mode, such as AEON.
        obstype (str): Observation type as in `OBSTYPE`
        slit (str): Value of `SLIT` keyword.
        grating (str): Value of `GRATING` keyword.
        wavmode (str): Value of `WAVMODE` keyword.
        roi (str): Value of `ROI` keyword.

    Returns:
        Observing technique as a string. Either `Imaging` or `Spectroscopy`.

    """
    if 'Spectroscopic' in roi or \
            obstype in ['ARC', 'SPECTRUM', 'COMP'] or \
            slit not in ['NO_MASK', '<NO MASK>'] or \
            grating not in ['NO_GRATING', '<NO GRATING>'] or \
            '_SP_' in target:
        technique = 'Spectroscopy'
    elif 'Imaging' in roi or \
            obstype in ['EXPOSE'] or\
            wavmode == 'IMAGING' or '_IM_' in target:
        technique = 'Imaging'
    else:
        technique = 'Unknown'
    return technique
```

In ``data_classifier.py``, on line 128 there is another reference but this might be for another thing. And it doesn't defines nothing. But it catches my attention that the line of self.technique is commented:

``` python
   def _get_obs_technique(self):
        """Identify if the data is Imaging or Spectroscopy

        For imaging data the keyword `WAVMODE` is `Imaging` therefore the logic
        here is: If there is only one value for `WAVMODE` and it is `Imaging`
        then the technique is `Imaging`. If `Imaging` is in the result along
        with other then it will assume the technique is Spectroscopy and will
        ignore all the Imaging data. If none of the conditions above are met it
        will assume the technique is Spectroscopy.

        The result is stored as an attribute of the class.

        """

        # self.technique = identify_technique()

        wavmodes = [str(w).upper() for w in self.objects_collection.wavmode.unique()]
        if len(wavmodes) == 1 and wavmodes[0] == 'IMAGING':
                self.technique = 'Imaging'

        elif 'IMAGING' in wavmodes and len(wavmodes) > 1:
                self.log.error('There seems to be Imaging and Spectroscopic '
                               'data. I will assume the Imaging data are '
                               'acquisition images therefore they will be '
                               'ignored.')
                self.log.info("If you really have Imaging data, please process "
                              "them in a separated folder.")
                self.technique = 'Spectroscopy'
        else:
                self.technique = 'Spectroscopy'
        # inform the results, no need to return
        self.log.info('Detected {:s} Data from {:s} '
                      'Camera'.format(self.technique, self.instrument))
```

Allonck, 2024-03-13:
I think this problem can be closed, from what I have been able to verify in the code

closed 2024-03-13 by Allonck