soar-telescope/goodman_pipeline #407: 'strptime' error

suman-phy, 2024-07-02:
I'm getting this error while running redccd.

File "XXX/goodman_pipeline/core/core.py", line 2271, in name_master_flats
    morning_twilight = datetime.datetime.strptime(morning_twilight,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: strptime() argument 1 must be str, not MaskedArray

simontorres, 2024-07-02:
I'm sorry you are having those issues. I'm suspecting there might be a version issue but before that I would like to be sure that you are using uncompressed files, my guess is that you are downloading your data from Las Cumbres?

If you are using uncompressed files (`*.fits` files) please copy and  paste the output of 
`pip freeze` from the environment where you are running the pipeline. 

This also applies for #406 

Simón.

suman-phy, 2024-07-02:
Hi. Thanks for your quick response. I'm using the data from observations made in the ongoing cycle from SOAR using Goodman. Yes, the data are in FITS format.

Below is the output from 'pip freeze'

`accessible-pygments==0.0.5
alabaster==0.7.16
asciitree==0.3.3
astroplan==0.10
astropy @ file:///croot/astropy_1717771811817/work
astropy-iers-data @ file:///croot/astropy-iers-data_1717700760580/work
astropy_healpix==1.0.3
astroscrappy==1.2.0
Babel==2.15.0
beautifulsoup4==4.12.3
Bottleneck @ file:///croot/bottleneck_1709069899917/work
ccdproc==2.4.2
certifi==2024.6.2
charset-normalizer==3.3.2
click==8.1.7
cloudpickle==3.0.0
contourpy @ file:///work/perseverance-python-buildout/croot/contourpy_1701756524386/work
coverage==7.5.4
coveralls==4.0.1
cycler @ file:///tmp/build/80754af9/cycler_1637851556182/work
Cython @ file:///croot/cython_1712282251484/work
dask==2024.6.2
docopt==0.6.2
docutils==0.21.2
fasteners==0.19
fonttools @ file:///croot/fonttools_1713551344105/work
fsspec==2024.6.1
goodman_pipeline==1.3.7
idna==3.7
imageio==2.34.2
imagesize==1.4.1
Jinja2==3.1.4
kiwisolver @ file:///work/perseverance-python-buildout/croot/kiwisolver_1698847502605/work
lazy_loader==0.4
locket==1.0.0
MarkupSafe==2.1.5
matplotlib @ file:///croot/matplotlib-suite_1713336378214/work
mkl-fft @ file:///work/perseverance-python-buildout/croot/mkl_fft_1698845673361/work
mkl-random @ file:///work/perseverance-python-buildout/croot/mkl_random_1698845720894/work
mkl-service==2.4.0
mock==5.1.0
networkx==3.3
numcodecs==0.12.1
numexpr @ file:///work/perseverance-python-buildout/croot/numexpr_1698871031164/work
numpy @ file:///croot/numpy_and_numpy_base_1708638617955/work/dist/numpy-1.26.4-cp312-cp312-linux_x86_64.whl#sha256=1d700f51d8b4fa684d858c9e3b56b1656bc5c82b6b79ff08d4e3b491c430059f
packaging @ file:///croot/packaging_1710807400464/work
pandas @ file:///croot/pandas_1718308974269/work/dist/pandas-2.2.2-cp312-cp312-linux_x86_64.whl#sha256=92c518f7e09edd50b5caa5862636c51d6a29391803f3ada62f68aa52f27d8f92
partd==1.4.2
pillow @ file:///croot/pillow_1714398848491/work
ply @ file:///work/perseverance-python-buildout/croot/ply_1698846188635/work
pockets==0.9.1
pydata-sphinx-theme==0.15.4
pyerfa @ file:///croot/pyerfa_1717700762709/work
Pygments==2.18.0
pyparsing @ file:///work/perseverance-python-buildout/croot/pyparsing_1698847881454/work
PyQt5==5.15.10
PyQt5-sip @ file:///work/perseverance-python-buildout/croot/pyqt-split_1698847927472/work/pyqt_sip
python-dateutil @ file:///croot/python-dateutil_1716495738603/work
pytz @ file:///croot/pytz_1713974312559/work
PyYAML @ file:///work/perseverance-python-buildout/croot/pyyaml_1698849903511/work
reproject==0.13.1
requests==2.32.3
scikit-image==0.24.0
scipy @ file:///croot/scipy_1717521478074/work/dist/scipy-1.13.1-cp312-cp312-linux_x86_64.whl#sha256=a60c1bdf983ee297b25e41830d00cc2af50f00bcb47e55578703fef1594ec56e
setuptools==69.5.1
sip @ file:///work/perseverance-python-buildout/croot/sip_1698846977033/work
six @ file:///tmp/build/80754af9/six_1644875935023/work
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.3.7
sphinxcontrib-applehelp==1.0.8
sphinxcontrib-devhelp==1.0.6
sphinxcontrib-htmlhelp==2.0.5
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-napoleon==0.7
sphinxcontrib-qthelp==1.0.7
sphinxcontrib-serializinghtml==1.1.10
tifffile==2024.6.18
toolz==0.12.1
tornado @ file:///croot/tornado_1718740109488/work
typing_extensions==4.12.2
tzdata @ file:///croot/python-tzdata_1690578112552/work
unicodedata2 @ file:///croot/unicodedata2_1713212950228/work
urllib3==2.2.2
wheel==0.43.0
zarr==2.18.2`

suman-phy, 2024-08-22:
Hi, can you please help me with this?

simontorres, 2024-08-22:
Hi, apologies for not working on this.

I did run checks, however I did not find any obvious errors, I even double checked the unit tests.

Now I'm exploring a new possible source.
Will let you know my findings

simontorres, 2024-09-09:
Hi @suman-phy I can not reproduce the error. Could you instead of pip freeze use `pip list --format=freeze`

Here is mine, in case you would like to try.
```
accessible-pygments==0.0.4
alabaster==0.7.13
appnope==0.1.4
asciitree==0.3.3
astroplan==0.9.1
astropy==5.3.4
astropy-healpix==1.0.0
astroscrappy==1.1.0
asttokens==2.4.1
attrs==24.2.0
Babel==2.13.0
backcall==0.2.0
backports.tarfile==1.1.1
beautifulsoup4==4.12.2
bleach==6.1.0
Bottleneck==1.3.7
build==1.2.1
ccdproc==2.4.1
certifi==2023.7.22
charset-normalizer==3.3.0
click==8.1.7
cloudpickle==2.2.1
contourpy==1.2.0
coverage==6.5.0
coveralls==3.3.1
cycler==0.11.0
Cython==3.0.10
dask==2023.9.3
decorator==5.1.1
defusedxml==0.7.1
docopt==0.6.2
docutils==0.20.1
entrypoints==0.4
executing==2.0.1
fasteners==0.19
fastjsonschema==2.20.0
fits-utilities==0.1.2
flake8==7.0.0
fonttools==4.51.0
fsspec==2023.9.2
goodman-focus==2.0.2
idna==3.4
imageio==2.31.5
imagesize==1.4.1
importlib-metadata==6.8.0
iniconfig==2.0.0
ipython==8.12.3
jaraco.classes==3.4.0
jaraco.context==5.3.0
jaraco.functools==4.0.1
jedi==0.19.1
Jinja2==3.1.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyterlab_pygments==0.3.0
keyring==25.2.1
kiwisolver==1.4.4
lazy_loader==0.3
locket==1.0.0
markdown-it-py==3.0.0
MarkupSafe==2.1.3
matplotlib==3.9.2
matplotlib-inline==0.1.6
mccabe==0.7.0
mdurl==0.1.2
mistune==3.0.2
mkl-fft==1.3.8
mkl-random==1.2.4
mkl-service==2.4.0
mock==5.1.0
more-itertools==10.2.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
networkx==3.1
nh3==0.2.17
numcodecs==0.11.0
numexpr==2.8.7
numpy==1.26.4
packaging==24.1
pandas==2.2.2
pandocfilters==1.5.1
parso==0.8.3
partd==1.4.1
pexpect==4.9.0
pickleshare==0.7.5
pillow==10.4.0
pip==24.1.2
pipreqs==0.5.0
pkginfo==1.10.0
platformdirs==4.3.2
pluggy==1.4.0
pockets==0.9.1
prompt-toolkit==3.0.43
ptyprocess==0.7.0
pure-eval==0.2.2
pycodestyle==2.11.1
pydata-sphinx-theme==0.14.1
pyerfa==2.0.0.3
pyflakes==3.2.0
Pygments==2.16.1
pyparsing==3.1.2
pyproject_hooks==1.1.0
PyQt5==5.15.10
PyQt5-Qt5==5.15.12
PyQt5-sip==12.13.0
pytest==8.1.1
python-dateutil==2.9.0.post0
pytz==2024.1
PyYAML==6.0.1
pyzmq==26.2.0
readme_renderer==43.0
referencing==0.35.1
reproject==0.12.0
requests==2.31.0
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.7.1
rpds-py==0.20.0
scikit-image==0.22.0
scipy==1.12.0
setuptools==72.1.0
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.2.6
sphinxcontrib-applehelp==1.0.7
sphinxcontrib-devhelp==1.0.5
sphinxcontrib-htmlhelp==2.0.4
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-napoleon==0.7
sphinxcontrib-qthelp==1.0.6
sphinxcontrib-serializinghtml==1.1.9
stack-data==0.6.3
tifffile==2023.9.26
tinycss2==1.3.0
toolz==0.12.0
tornado==6.4.1
traitlets==5.14.1
twine==5.0.0
typing_extensions==4.8.0
tzdata==2023.3
unicodedata2==15.1.0
urllib3==2.0.6
validate-pyproject==0.18
wcwidth==0.2.13
webencodings==0.5.1
wheel==0.43.0
yarg==0.1.9
zarr==2.16.1
zipp==3.17.0
```

suman-phy, 2024-09-10:
Hi @simontorres, below is the list. The versions of the packages are much newer here, as I had recently updated the server. I think the errors are a result of some changes in the updated packages. Can you please update your installation and test?

accessible-pygments==0.0.5
alabaster==1.0.0
asciitree==0.3.3
astroplan==0.10.1
astropy==6.1.3
astropy_healpix==1.0.3
astropy-iers-data==0.2024.9.2.0.33.23
astroscrappy==1.2.0
autocommand==2.2.2
babel==2.16.0
backports.tarfile==1.2.0
beautifulsoup4==4.12.3
Bottleneck==1.3.7
ccdproc==2.4.2
certifi==2024.8.30
charset-normalizer==3.3.2
click==8.1.7
cloudpickle==3.0.0
contourpy==1.2.0
coverage==7.6.1
coveralls==4.0.1
cycler==0.11.0
Cython==3.0.10
dask==2024.8.2
docopt==0.6.2
docutils==0.21.2
fasteners==0.19
fonttools==4.51.0
fsspec==2024.9.0
goodman_pipeline==1.3.8
goodman_pipeline==1.3.8
idna==3.8
imageio==2.35.1
imagesize==1.4.1
importlib_metadata==8.0.0
importlib_resources==6.4.0
inflect==7.3.1
jaraco.context==5.3.0
jaraco.functools==4.0.1
jaraco.text==3.12.1
Jinja2==3.1.4
kiwisolver==1.4.4
lazy_loader==0.4
locket==1.0.0
MarkupSafe==2.1.5
matplotlib==3.9.2
mkl_fft==1.3.10
mkl_random==1.2.7
mkl-service==2.4.0
mock==5.1.0
more-itertools==10.3.0
networkx==3.3
numcodecs==0.13.0
numexpr==2.8.7
numpy==1.26.4
ordered-set==4.1.0
packaging==24.1
pandas==2.2.2
partd==1.4.2
pillow==10.4.0
pip==24.2
platformdirs==4.2.2
ply==3.11
pockets==0.9.1
pydata-sphinx-theme==0.15.4
pyerfa==2.0.1.4
Pygments==2.18.0
pyparsing==3.1.2
PyQt5==5.15.10
PyQt5-sip==12.13.0
python-dateutil==2.9.0.post0
pytz==2024.1
PyYAML==6.0.2
reproject==0.14.0
requests==2.32.3
scikit-image==0.24.0
scipy==1.13.1
setuptools==72.1.0
sip==6.7.12
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.6
Sphinx==8.0.2
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-napoleon==0.7
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
tifffile==2024.8.30
tomli==2.0.1
toolz==0.12.1
tornado==6.4.1
typeguard==4.3.0
typing_extensions==4.12.2
tzdata==2023.3
unicodedata2==15.1.0
urllib3==2.2.2
wheel==0.43.0
zarr==2.18.3
zipp==3.19.2

simontorres, 2024-12-17:
The problem seem to happen in astroplan, issue created https://github.com/astropy/astroplan/issues/608